OctoPrint/OctoPrint-FirmwareUpdater #1: Velleman Vertex K8400 not supported

FredrikWendt, 2016-09-29:
I have a Velleman Vertex 3D K8400 printer and I could not use the "Check for Updates" button. I think this should be fixed and I'm happy to hack away at this if someone could just point me in the right direction.

nicanor-romero, 2016-09-29:
Hi @FredrikWendt , once installed you have to edit the `config.yaml` file with the URL in which you will check for updates. You also need to make sure your printer returns the MACHINE_TYPE and FIRMWARE_VERSION as a M115's response. This information is written in the README.md file.

Greetings!

FredrikWendt, 2016-10-02:
There is no official pre-built binary for Velleman's 3D printer. The only supported upstream method to build the firmware is using the Arduino IDE, since the only supported Arduino version is a very old one.

closed 2016-10-02 by FredrikWendt