OctoPrint/OctoPrint-FirmwareUpdater #3: plugin not in plugin repository

mtowara, 2016-11-13:
Hi,
this is really super useful, stumbeld on it by accident. Why is this not in the official plugin repository (thus your README suggests it should be)?

Cheers,
Markus

danekshea, 2016-12-13:
Hey! Did it work for you mtowara? If so, it is a shame that this isn't in the official repository! SO useful!

foosel, 2016-12-14:
It's currently not in the repository since right not it also contains an undocumented API for fetching notifications about available firmware updates from vendors (which either needs to be documented or removed) and also because I'm not sure if @nicanor-romero (who's the original author) is still maintaining the general available version. I'll happily "inherit" it for general consumption, but I need to know the state of this ;)

SCiunczyk, 2017-02-01:
Seams that plug-in is not maintained any more :-(

foosel, 2017-02-01:
Indeed. As written earlier, I'll happily adopt it, but it needs some work to get rid of the aforementioned undocumented bits (or better replacing them) and I'll have to see when I can find the time for that.

danekshea, 2017-02-01:
I currently use this plugin in my business Gina and just like the history
plugin that is currently being further developed, I'll happily pay a
freelancer to do the things that are needed for this plugin to meet your
requirments as long as it's not too expensive. Can you be more specific on
exactly what needs to be done so I can get a quote and the freelancer knows
exactly what to do.

Thanks,

Dane

Erutan409, 2017-02-11:
@foosel Pure curiosity - would you be more inclined to adopt it if it weren't for the auto-update functionality that calls out to an undocumented API?  I used this for my Robo 3D R1+ and I just pointed it to the direct link the manufacturer had on their website.  It appeared to work for me.  But, when attempting to use their API, nothing was found.  So, for me, the aforementioned functionality was completely useless.  But, not having to open up my printer (with the way I have my raspberry pi installed) and using OctoPrint to update my firmware is MUCH preferred; if I'm trusting its stability.

foosel, 2017-02-13:
I've already started stripping the undocumented parts out, so I'm more or less settled on adopting it.

danekshea, 2017-02-13:
The workload you have taken on is crazy... We all thank you very, very much.

SCiunczyk, 2017-02-13:
Thanks Gina

foosel, 2017-02-13:
My hope with this particular part of my workload is that once the API access stuff is stripped the work to maintain this plugin decreases drastically ;) If that turns out to be a misconception, I'll have to give this up for adoption after all because OctoPrint itself is already taking up pretty much all of my available time. But let's try and see :)

danekshea, 2017-02-13:
I have a financial and personal interest in keeping this plugin going so if
I can help in any way, let me know! I'll hire somebody to clean up some
stuff if necessary!

SCiunczyk, 2017-02-13:
I can offer my help with tests of course...

benlye, 2017-07-10:
Is anyone still maintaining or working on this?  Is it worth submitting PRs for changes/enhancements against this repository, or should it be spun off into a new plugin?

benlye, 2017-07-11:
@foosel, if you want a volunteer to do some work on this plugin to strip out the API updater stuff, I'm happy to have a go at it.

foosel, 2017-07-11:
@benlye thanks for the offer! I actually do have something on my disk that is (hopefully O_O) the plugin minus the API updater stuff, I just still haven't gotten around to actually testing it. Might it be possible that I push it into a branch and you give it a test whirl and potentially iron out what's broken?

I sadly had even less time for plugin maintenance than I anticipated thanks to the workload of OctoPrint itself :/

benlye, 2017-07-11:
> Might it be possible that I push it into a branch and you give it a test whirl and potentially iron out what's broken?

Yes, of course!  I've been working on it a little bit on my own and have a for where I've added configuration settings for the MCU type and programmer, so that it will work with my Atmega1284p board.  I can contribute that back to your new branch as well.

foosel, 2017-07-12:
Perfect! See the `devel` branch I just pushed, commit is ea292e4:

> Heavy refactoring to make ready for adding to repo
>
>  * Removed undocumented update API stuff
>  * First steps towards supporting more than just avrdude + wiring +
>    mega2560 configs
>  * Messages from backend can now be translated through the use of
>    message types and subtypes instead of full blown strings.
>  * Introduced console log
>  * Fixed python setup.py sdist
>  * Incremented version number
>
> TODO: Tests need to be either fixed or removed, as of now they are
> utterly and completely broken due to the refactoring.

A first test against my printer here looked good - it was still flashing an uploaded hex file, so I didn't break everything it seems ;) Still, more tests are more than welcome.

benlye, 2017-07-12:
I'll take a look :-)

Quas7, 2017-07-29:
Tested the flash plugin with Mega2560 Trigorilla mainboard on my Anycubic prusa i3 using a Octopi on Raspberry Pi 3.

Exported .hex without bootloader via Arduino IDE --> Sketch --> export binary for this purpose.
Disconnected Octoprint, flashed and it worked without any issues.

Nice work!

foosel, 2017-08-23:
@Quas7 thanks for the feedback! Sounds good so far... @benlye Did you have a chance to take a look as well? (Also, yes, I know, I still have to take a look at your PR here - I just came back from an overdue vacation and have been busy wading through a metric ton of email ;))

motey, 2017-08-23:
Tested devel branch on Octopi@Raspberry Pi 3 with a [MKS Board V1.4 ](http://reprap.org/wiki/MKS_GEN) (ATmega2560)
and Marlin 1.4.
I used .hex **with** bootloader. (not sure which export is more correct. anybody informations about this?)

Nice Work!

benlye, 2017-08-23:
@foosel I've been using this version with my changes to allow different AVR MCUs and add backing up and it works well.  I've flashed the firmware on both my Atmega1284p and RAMPS 2560 many times now using it with no issues at all - I'm very happy not to have to mess around with USB cables whenever I want to update my firmware.  The only thing I've observed is that the status bar goes to 'writing firmware' but never changes to 'verifying'.  This doesn't affect the functionality at all, and I've not had time to take a look into it myself.

andrivet, 2017-08-28:
For info: I have just used this plug-in (dev branch) to update my Wanhao i3 Plus (Marlin) to the latest version (v3.0) and it works perfectly.

th33xitus, 2017-09-01:
Tested the devel branch on my Raspi Zero W with an Anet A8 Board V1.0 with an ATmega1284P for flashing Marlin 1.1.5.

I tried both the Marlin.ino.sanguino.hex and Marlin.ino.with_bootloader.sanguino.hex i successfully compiled with Arduino IDE 1.8.0 but neither of them was flashed.
The problem i have is, that i am getting a "Flashing Failed - Timeout communicating with programmer" message after a short while.

AVRDUDE was installed of course and i specified the path in the plugins settings. Also the serialport is set correctly to /dev/ttyUSB0

Am i missing something? Is it the Zero W which is failing there? Actually all other functions of Octoprint works really fine with this Pi and there are no issues at all controlling my printer.

benlye, 2017-09-01:
@th33xitus, the version here only works with Atmega2560 based boards. For the Anet you need to try my repo, which lets you select the 1284p. Would be good to know if it works for you too (it works for me). 

https://github.com/benlye/OctoPrint-FirmwareUpdater/tree/benlye-devel

th33xitus, 2017-09-01:
@benlye Hey thank you for that advice! Sadly it doesn't work as well. The message i receive is now "Flashing failed - Unknown error" after the progress bar at the bottom showed "Starting flash" for around 10-20 seconds.

I left the path to avrdude config blank, because i don't know if and where it is located on my Zero W.
AVR MCU Type was set to Atmega1284p of course and i tried both AVR Programmer Types you can select from the dropdown menu.

benlye, 2017-09-01:
The OctoPrint log should tell you exactly what is going wrong. My guess is it will be path related. To find Avrdude, log onto your pi and run `which avrdude`.

th33xitus, 2017-09-01:
@Quas7 
No i always make sure that the printer is connected before i hit the flash button. But as soon as i hit the button, the printer will be disconnected. The progress bar starts with "Flashing firmware..." then follows a quick "Disconnecting printer"... followed by "Start flashing..." then it takes 10 seconds and i receive the "Flashing failed - Unknown error" message

I looked into the logfiles like @benlye told me to and before that i ran:
`
pi@octopi:~ $ which avrdude
/usr/bin/avrdude
pi@octopi:~ $
`

Here is the octoprint.log:
`
2017-09-01 12:10:04,953 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.5 (Github)"
2017-09-01 12:10:09,907 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.7
2017-09-01 12:10:09,910 - octoprint.server.util.sockjs - WARNING - Could not send message to client 192.168.1.7: object of type 'NoneType' has no len()
2017-09-01 12:10:15,937 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.7
2017-09-01 12:10:19,934 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.7
2017-09-01 12:10:19,939 - octoprint.server.util.sockjs - WARNING - Could not send message to client 192.168.1.7: object of type 'NoneType' has no len()
2017-09-01 12:10:25,932 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.7
2017-09-01 12:10:29,951 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.7
2017-09-01 12:10:29,956 - octoprint.server.util.sockjs - WARNING - Could not send message to client 192.168.1.7: object of type 'NoneType' has no len()
2017-09-01 12:10:30,221 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2017-09-01 12:10:30,368 - octoprint.plugins.firmwareupdater - INFO - Running u'/usr/bin/avrdude -v -p m1284p -c arduino -P /dev/ttyUSB0 -U flash:w:/tmp/tmpG18s4j:i -D' in /usr/bin
2017-09-01 12:10:34,077 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude returned code 1.
2017-09-01 12:10:34,128 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'name': 'Anet A8', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 300, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 220.0, 'width': 220.0, 'custom_box': False, 'height': 240.0}, 'model': 'Anet A8', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2017-09-01 12:10:34,646 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-01 12:10:35,918 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.7
`
I don't know if all those info entrys about "new connection .. " and " ... connection closed" are normal?
But the log also says, that the printer gets disconnected, then it runs avrdude (which is located in the proper directory i assume? the testing function of the plugin also says that the path is valid), fails due to code 1 (whatever that means) and then reconnecting to my printer.

benlye, 2017-09-01:
You could try manually copying the hex file to somewhere else on the Pi and running the same avrdude command manually. You should get some verbose output. The port looks OK so I wonder if it doesn't like your Baud rate. Which bootloader do got have on the Anet board? Optiboot or standard?

th33xitus, 2017-09-01:
via google i took some information from a german forum and created a script with the following content:
`avrdude -C /etc/avrdude.conf -v -p m1284p -c wiring -P /dev/ttyUSB0 -b 115200 -D -U flash:w:./Marlin.ino.with_bootloader.sanguino.hex:i`

i execute the script with `sudo /home/pi/flash.sh`. the hex and script file are in the same directory.
now as you can see i even included the baudrate my pi is communicating with the printer usually.
but i still get timeout messages:
> avrdude: Version 6.1, compiled on Jul  7 2015 at 10:29:47
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/etc/avrdude.conf"
>          User configuration file is "/root/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : /dev/ttyUSB0
>          Using Programmer              : wiring
>          Overriding Baud Rate          : 115200
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_getsync(): timeout communicating with programmer
> 
> avrdude done.  Thank you.

to answer your question about the bootloader im missing knowledge in this topic. i can't answer that question. but i could provide the data of the hardware folder from my arduino IDE for the anet board if you tell me how i can attach it here (github tells me that it cant process my zipfile with the data in it -.- )

benlye, 2017-09-01:
Try again with `-c arduino`, which is the correct programmer for this board. Another approach is to set your Arduino IDE to provide verbose output then flash the board using that. You will be able to see the avrdude command line it uses and replicate it on the Pi command line.

th33xitus, 2017-09-01:
When i try it with -c arduino i get nearly the same output.

> avrdude: Version 6.1, compiled on Jul  7 2015 at 10:29:47
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/etc/avrdude.conf"
>          User configuration file is "/root/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : /dev/ttyUSB0
>          Using Programmer              : arduino
>          Overriding Baud Rate          : 115200
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x73
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x74
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x61
> avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
> avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x74
> avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x0a
> avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x65
> avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x63
> avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x68
> avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6f
> 
> avrdude done.  Thank you.

But holy fuck! I love you for the idea with just looking up what my arduino ide says. We are getting closer.
I changed the command to:
`avrdude -C/etc/avrdude.conf -v -v -v -patmega1284p -carduino -P/dev/ttyUSB0 -b57600 -D -Uflash:w:./Marlin.ino.with_bootloader.sanguino.hex:i`
and now i get at least some responses of avrdude:

> pi@octopi:~ $ sudo /home/pi/flash.sh
> ● octoprint.service - LSB: OctoPrint daemon
>    Loaded: loaded (/etc/init.d/octoprint)
>    Active: inactive (dead) since Fri 2017-09-01 15:25:59 UTC; 2min 36s ago
>   Process: 2575 ExecStop=/etc/init.d/octoprint stop (code=exited, status=0/SUCCESS)
>   Process: 2481 ExecStart=/etc/init.d/octoprint start (code=exited, status=0/SUCCESS)
> 
> Sep 01 15:22:41 octopi octoprint[2481]: Starting OctoPrint Daemon: OctoPrint.
> Sep 01 15:22:41 octopi systemd[1]: Started LSB: OctoPrint daemon.
> Sep 01 15:25:58 octopi systemd[1]: Stopping LSB: OctoPrint daemon...
> Sep 01 15:25:59 octopi octoprint[2575]: Stopping OctoPrint Daemon: OctoPrint.
> Sep 01 15:25:59 octopi systemd[1]: Stopped LSB: OctoPrint daemon.
> Sep 01 15:28:35 octopi systemd[1]: Stopped LSB: OctoPrint daemon.
> 
> avrdude: Version 6.1, compiled on Jul  7 2015 at 10:29:47
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/etc/avrdude.conf"
>          User configuration file is "/root/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : /dev/ttyUSB0
>          Using Programmer              : arduino
>          Overriding Baud Rate          : 57600
>          AVR Part                      : ATmega1284P
>          Chip Erase delay              : 55000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
[... 57 lines cut ...]
> avrdude: safemode: hfuse reads as 0
> avrdude: safemode read 1, efuse value: 0
> avrdude: safemode read 2, efuse value: 0
> avrdude: safemode read 3, efuse value: 0
> avrdude: safemode: efuse reads as 0
> avrdude: reading input file "./Marlin.ino.with_bootloader.sanguino.hex"
> avrdude: writing flash (128986 bytes):
> 
> Writing |                                                    | 0% 0.00s
> avrdude: stk500_paged_write(): (a) protocol error, expect=0x14, resp=0x21
> avrdude: stk500_cmd(): programmer is out of sync
> ● octoprint.service - LSB: OctoPrint daemon
>    Loaded: loaded (/etc/init.d/octoprint)
>    Active: active (running) since Fri 2017-09-01 15:28:38 UTC; 162ms ago
>   Process: 2575 ExecStop=/etc/init.d/octoprint stop (code=exited, status=0/SUCCESS)
>   Process: 2650 ExecStart=/etc/init.d/octoprint start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/octoprint.service
>            └─2656 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --po...
> 
> Sep 01 15:28:38 octopi octoprint[2650]: Starting OctoPrint Daemon: OctoPrint.
> Sep 01 15:28:38 octopi systemd[1]: Started LSB: OctoPrint daemon.
> pi@octopi:~ $

As you can see writing still fails but at least there is apparently some kind of communication between the programmer on my pi and my printer.
Any clue now how i can manage to write the hex file via putty and AFTER that realise it with the plugin? :-/

b-morgan, 2017-10-26:
Are we getting any closer to having this plugin officially supported? I would really love to keep my printer connected to OctoPrint all the time. 

The beta test of Cura2 Lulzbot Edition is currently changing the firmware almost daily and moving the USB cable back and forth is getting old!

Thanks!

foosel, 2017-10-27:
I want to get a feature in that @benlye wanted to send a separate PR for and am still waiting for that (see the discussion in #11).

Until then you can just manually install from https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/devel.zip

n3qik, 2017-12-03:
Have the below working on my Anet A8 stock board.
avrdude -C /etc/avrdude.conf -v -p m1284p -c arduino -P/dev/ttyUSB0 -b 57600 -D -U flash:w:./3dprinter/A8-S_firmwareUpdate.hex:i

/3dprinter is the shared folder i have on the pi.
Using this: https://github.com/benlye/OctoPrint-FirmwareUpdater/tree/benlye-devel

benlye, 2018-01-05:
@foosel I've updated the master branch with the changes to allow MCU type and programmer - the stripped down version without backup, thanks @andrivet.  I've fixed the issue in #6, updated the readme, and made a few other minor tweaks.  

I think we're good to go 😄

benlye, 2018-01-15:
Aaaand, we're in the plug-in repository - thanks, Gina 😄 

https://plugins.octoprint.org/plugins/firmwareupdater/

closed 2018-01-15 by benlye

foosel, 2018-01-15:
:tada: Thanks for taking care of this, having this on good hands really put a load off my shoulders :smile:

andrivet, 2018-01-15:
This is great!
Thank you Gina and Ben.
Sebastien

https://plugins.octoprint.org/plugins/firmwareupdater/