OctoPrint/OctoPrint-FirmwareUpdater #7: RFE: post-update g-code

Andrei-Pozolotin, 2017-06-30:
normally there is a need to run small g-code snippet right after update
this request if to provide extra editable area in UI for such g-code

benlye, 2018-01-21:
@Andrei-Pozolotin, are you still interested in this feature?  I've added it in this branch:
https://github.com/benlye/OctoPrint-FirmwareUpdater/tree/postflash

You can manually install that version from here:
https://github.com/benlye/OctoPrint-FirmwareUpdater/archive/postflash.zip

[image]

How it works:
1. The updater plugin registers an event handler and watches for a `CONNECTED` event
1. If post-flash gcode is enabled and there are commands set, after firmware is flashed successfully the updater plugin sets a flag indicating that post-flash code needs to be run
1. When the post flash flag is set and the `CONNECTED` event is received by the plugin the post-flash command(s) are sent to the printer

The post-flash code will run more or less immediately if the printer was connected before the flash started (so reconnects automatically when the flash finishes), or whenever the printer is manually reconnected.

It should be ready to go, but would be nice to have another tester before I merge it.

Andrei-Pozolotin, 2018-01-21:
great, thank you, will take a look

benlye, 2018-01-24:
I've spent a few days testing so this should be good to go - I went ahead and merged it into the main repo.  

I'm already finding it useful - my post flash code is `M502; M851 Z-4; M500; M117 Firmware flashed`.

closed 2018-01-24 by benlye