OctoPrint/OctoPrint-FirmwareUpdater #8: FirmwareUpdater not dettecting the firmware

MiquelAdell, 2017-07-07:
Hi!

I've got a printer from a friend and I was trying to update it.

I set up an octoprint for it and it seems to be working. 

The FirmwareUpdater is returning "Unable to get printer's data" though.

If  I send `M115` on the terminal I get:

```
Recv: FIRMWARE_NAME: Marlin 1.1.0-RC2 From Archive 
SOURCE_CODE_URL: https://github.com/fmalpartida/Marlin/tree/SAV-MkI-config 
PROTOCOL_VERSION: 1.0 
MACHINE_TYPE: SAV MkI 
EXTRUDER_COUNT: 1 
UUID: 00000000-0000-0000-0000-000000000000
```

which seems ok.

The issue will be FirmwareUpdater not dettecting current hardware or machine but I was not expecting that to work with a 2+ year old firmware.

I can provide some help debugging that or, otherwise, I would be happy to skip the detection step if you could provide me some guidance on how to query the web service myself so I can get the hex file and upload it.
(I'm assuming the plugin queries the web service with some information from the printer and it gets a compiled hex file back for the current printer. Please correct me if I'm wrong).

I would prefer not to need to compile the firmware if I can avoid it (and of course that would not be an issue for this plugin).

Thanks!

foosel, 2017-07-12:
That bit which it needs to detect the firmware for in the first place is part of a vendor specific update mechanism that will be stripped from the next version anyhow. I can't tell you how to query/setup a webservice for that update mechanism since I don't know it either - which is why this part is going to get stripped.

Versions going forward will require you to upload the firmware hex file, either as file or from an URL. You won't have to compile it yourself, if your printer vendor provides it that should work too, but some kind of update detection is off the table for the foreseeable future simply because it would require external infrastructure to maintain which takes time at least I do not have :)

closed 2017-07-12 by foosel

MiquelAdell, 2017-07-12:
Thanks! don't worry I'm very happy to see that the plugin is being maintained anyway