OctoPrint/OctoPrint-FirmwareUpdater #9: Allow different MCU and programer

benlye, 2017-07-10:
I was able to use the plugin to flash a new .hex file to my Atmega1284p board, but only after changing the avrdude options at https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/octoprint_firmwareupdater/__init__.py#L179:

`avrdude_command = [avrdude_path, "-v", "-p", "m1284p", "-c", "arduino", "-P", selected_port, "-U", "flash:w:" + hex_path + ":i", "-D"]`

It would be nice to allow the user to pick the MCU part and programmer from a dropdown, or allow the avrdude command line to be edited in the settings.

foosel, 2017-07-12:
I think it would be even better to take this one step further and allow alternative flashing methods (not ever printer is powered by an AVR) through a custom plugin hook.

benlye, 2017-07-12:
> I think it would be even better to take this one step further and allow alternative flashing methods (not ever printer is powered by an AVR) through a custom plugin hook.

OK, I need to do some reading to work out what that means ;-)

I guess my attempt was somewhat more basic, but does work:
[image]

benlye, 2017-07-24:
It's been a while so I just wanted to say I'm still working on this.   

As well as extending the configuration options to allow different MCUs/programmers, I'm adding a 'back up' button to read the current firmware and send it back to the client.

I only have AVR hardware, so I'm just focussed on the avrdude part rather than extending towards other hardware types.

foosel, 2017-07-25:
> As well as extending the configuration options to allow different MCUs/programmers, I'm adding a 'back up' button to read the current firmware and send it back to the client.

Oooooh! That's an awesome idea! 😍👍

> I only have AVR hardware, so I'm just focussed on the avrdude part rather than extending towards other hardware types.

Totally fine as a first throw IMHO :) Just wanted to mention it because that's something that has been floating around in my head from time to time ;)

closed 2018-01-05 by benlye