OctoPrint/OctoPrint-FirmwareUpdater #10: Breaks Latest Octoprint

sashasklar, 2017-07-23:
Installed on latest Octoprint image, no other plugins except TouchUI. After installing FirmwareUpdater, can no longer connect to printer.

foosel, 2017-07-24:
Cannot reproduce, please provide logs.

ManuGithubSteam, 2017-07-28:
I can not confirm this with a Prusa i2 Mk2

sashasklar, 2017-07-29:
Sorry for the lag in response, here's some more details:

OctoPrint: 1.3.5rc2  
Prusa i3 Mk: Firmware 3.0.11

Terminal Output:
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x71fe19f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=90.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Recv: start
Send: N0 M110 N0*125
Recv: echo:echo: Last Updated: Feb 14 2017 15:35:19 | Author: (none, default config)
Recv: Compiled: Feb 14 2017
Recv: echo: Free Memory: 2670  PlannerBufferBytes: 1280
Recv: echo:Stored settings retrieved
Recv: echo:SD init fail
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor

sashasklar, 2017-08-01:
After digging into this further, it turns out the issue was on the printer side. The z-tuning message on the Prusa MK2, which is cleared by running the V2Calibration.gcode or sending the prusa-specific G87 code, was preventing Octoprint from connecting.

closed 2017-08-01 by sashasklar