OctoPrint/OctoPrint-FirmwareUpdater #21: PRUSA MK2/S USB RAMBO MINI and MK3 EINSY Support

TheZeroBeast, 2018-01-16:
Is there a way we can add options in the upload section to support the PRUSA printers both via USB and the EINSY second serial port for Octopi.

Much appreciated

Rob

benlye, 2018-01-16:
I don't have one of those and don't know anything about them.  How do you normally update the firmware?  Is it done with the Arduino IDE?  Can you point me to any documentation?

TheZeroBeast, 2018-01-16:
Hi Ben,

Arduino IDE as per the libraries and code in the below github.

https://github.com/prusa3d/Prusa-Firmware

Regards,
Robert McKenzie

benlye, 2018-01-16:
Documentation's a bit vague, isn't it?

Looking through the boards.txt file it seems like the RAMbo is an Atmega2560, which means you should be able to flash it by selecting `Atmega2560` as the MCU and `wiring` as the programmer.

Not sure about the second serial port - does that not show up in the list of ports in the drop down?

TheZeroBeast, 2018-01-17:
Hi Ben,

I can confirm that the Atmega2560 and wiring combination worked a treat, sorry I didn't try that before making this.

Rob

closed 2018-01-17 by TheZeroBeast

benlye, 2018-01-17:
😃