OctoPrint/OctoPrint-FirmwareUpdater #22: Howto Uninstall in terminal

carlymx, 2018-01-16:
tnk

benlye, 2018-01-16:
Not sure I follow - you want to remove the plug-in via the Linux shell?  Why not use the plugin manager?

benlye, 2018-01-16:
You can either run the command that the plugin manager would run:
`~/OctoPrint/venv/bin/python -m pip uninstall --yes OctoPrint-FirmwareUpdater`

or remove the plugin files manually:
```
rm -rf ~/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_firmwareupdater/
rm -rf ~/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint_FirmwareUpdater-0.2.0-py2.7.egg-info/
```
And then restart OctoPrint.

carlymx, 2018-01-17:
Hello,

I have started a project of a new distribution to use an Orange Pi Zero as a print server with Octoprint called OrangePrint, as well as OctoPi but for the orange machine.

I found this plugin very interesting and decided to try it to include it in the Distro, I tried to install it by the command line with the command:

```
sudo apt-get update
sudo apt-get install avrdude
pip install "https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/archive/master.zip"
(Described on its website: https://plugins.octoprint.org/plugins/firmwareupdater/
```

and the result was after several restarts and updates of the explorer cache what you can see in image:

[image]

The plugin did not appear in the corresponding menu and the option to uninstall it was disabled.

That's why I asked for the command to uninstall by terminal.

Using the installer that brings the Octoprint, where they leave the list of available plugins to operated in appearance without any problem.

Maybe they should look at it in case it's a bug.

Greetings and thank you.

benlye, 2018-01-17:
OK.  I just tested and can install and remove the plugin via the command line pip command, so I'm not sure it's a bug in the plugin installer.  You must use the pip instance in the OctoPrint environment.

closed 2018-01-17 by benlye