OctoPrint/OctoPrint-FirmwareUpdater #23: Flashing while still connected gives 'Unknown Error'

keithejc, 2018-01-18:
Flashing while Octoprint is connected to the printer will fail and the error message is 'Unknown Error'.
Can the plugin detect if the printer is connected and warn the user with a more friendly message?

benlye, 2018-01-18:
I flash while connected all the time and haven't run into that problem.  The plugin detects the active connection and disconnects the printer before starting.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/octoprint_firmwareupdater/__init__.py#L172

Are you using the same serial port for OctoPrint communication and flashing?

benlye, 2018-01-18:
Is OctoPrint configured to use 'AUTO' instead of a specific port on communicate?  I can see how that might be a problem in the connection check.  I can't check now but will look into later if you can confirm your setup.

benlye, 2018-01-18:
Yep, it seems to behave as you describe if the OctoPrint serial port is set to 'AUTO'.  I'll work out how to fix that, but in the mean time if you change your OctoPrint config to use a specific serial port you shouldn't have that problem anymore.

benlye, 2018-01-18:
Fixed by https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/24/commits/dd7739e3ef7d2a0f3dff4992a0932877608f2feb in #24.

closed 2018-01-18 by benlye