OctoPrint/OctoPrint-FirmwareUpdater #26: Bootloader files

Alissonverd, 2018-01-25:
Block files with bootloader in name. 
I tried sending a firmware with bootloader and it corrupted my card. After that the only solution was to write the bootloader again with another arduino.
Files without the bootloader work perfectly!

Alissonverd, 2018-01-25:
To reflash the bootloader I use this tutorial: https://learn.sparkfun.com/tutorials/installing-an-arduino-bootloader
and
https://www.arduino.cc/en/Tutorial/ArduinoISP

Alissonverd, 2018-01-25:
And thank you for this amazing plugin!

Alissonverd, 2018-01-25:
Take a look at this, I do not know if it helps.

Marlin.ino.mega.hex
End of file
`:101A4C00002F00537465707261746520746F6F2081
:101A5C00686967683A20005A00590058004361735E
:101A6C0065206C696768743A206F6E00436173651A
:101A7C00206C696768743A206F666600646973004D
:101A8C004D313038004D313132004D3431300000A1
:00000001FF`

Marlin.ino.with_bootloader.mega.hex
End of file
`:10FCE0005A95A9F780959095BC01CD010895F99991
:10FCF000FECF92BD81BDF89A992780B50895262F31
:10FD0000F999FECF1FBA92BD81BD20BD0FB6F89400
:0EFD1000FA9AF99A0FBE01960895F894FFCF63
:040000033000E000E9
:00000001FF`

Alissonverd, 2018-01-25:
Check file for this two keys:
426F6F746C6F61646572 < Bootloader
41546D65676132353630 < ATmega2560

benlye, 2018-01-26:
I can see another way to do this, which wouldn't require decoding the hex data...

We know which MCU the user has, and the datasheets tell us the start addresses of the flash bootloader section for each MCU we support, so we can just look for any matching lines.

For example, an Atmega2560 supports 512, 1024, 2048, or 4096 word bootloaders, with 4096 being the default.  The start address for each of those is 0x3FC00, 0x3F800, 0x3F000, and 0x3E000, with 0x3E000 being the default.

In the Atmega2560 file with the bootloader we see this line:
`:020000023000CC`

Which instructs that each subsequent data record should be offset by `0x3000 * 16`.  So, the following line:
`:10E000000D9489F10D94B2F10D94B2F10D94B2F129`
is really being written at `(0x3000 * 16) + 0xE000`, which is 0x3E000, one of the known bootloader start addresses for the Atmega2560.

I'll work on a way to implement this check and throw a warning.  It's should be fairly straight-forward.

Alissonverd, 2018-01-26:
=D
I have basic programming skills. So I can only get ideas and I can not implement them.
Many thanks for your support!

codecvlt, 2018-01-29:
+1 Totally bricked my cr-10s because of this. Ordered an Arduino uno so i can reflash the bootloader. The fun never ends :)

benlye, 2018-01-29:
I haven't had time to implement the check yet, so I've added a warning (plus a few instructions) to the firmware update page.  Hopefully this helps...

[image]

benlye, 2018-03-13:
I've released v1.0.0 which will warn if 'bootloader' is in the hex file name.

closed 2018-03-13 by benlye