OctoPrint/OctoPrint-FirmwareUpdater #27: Disconnect the printer or dont send the firmware

Alissonverd, 2018-01-25:
Hello,
We need to add a printer connection check before sending the firmware. 
If the printer is connected or the USB is busy, we get an error message when sending the firmware and after difficulties in disconnecting and reconnecting.

Thank you

benlye, 2018-01-25:
Please explain further. 

If the printer is connected it will be disconnected automatically. It's always worked for me.

Can you attach the relevant part of the OctoPrint log?

Are you running the latest version, 0.4.0? The initial release had an issue when the connection was set to AUTO.

Alissonverd, 2018-01-25:
[image]
If the printer in "State: Operational" and you send the firmware.. we have this msg.. and problems after it.

benlye, 2018-01-25:
Please update to the latest version.

Alissonverd, 2018-01-25:
I will install the last version and test. I dont receive any order to update the plugin.

Alissonverd, 2018-01-25:
So. See the problem with bootloader files.. 
It took my sleep for a whole night. haha

benlye, 2018-01-25:
Yes, I saw that too. Still  thinking about how to address it... Not sure what I can do other than adding a warning not to flash files that include a bootloader.

Alissonverd, 2018-01-25:
Maybe only check the file name in first time and stop proccess.

Alissonverd, 2018-01-25:
The new version working fine. Disconnect, flash and reconnect!
I will use the post-flash gcode.

Sorry for dont check the version.
Amazing work! thank you!

benlye, 2018-01-25:
:-)

closed 2018-01-25 by benlye