OctoPrint/OctoPrint-FirmwareUpdater #30: Get unknown error on every FW Update

riisefrutti, 2018-01-30:
When I try to update fw with the FirmwareUpdater I get through the flashing, but when it is supposed to reset I get FWUpdate fail - unknown error. 

It looks like the FW is flashed to the tarantula, and it works great. Cant wrap my head around why the error occures.

benlye, 2018-01-30:
What version of the plugin are you using?  What does the octoprint.log log file show?

riisefrutti, 2018-01-30:
I have tried both 0.4 and 0.5. 

The FW works flawlessly even with the update error.

Ill post from the log When I Get home again.

Aleks

Sendt fra min iPhone

grandmastermp, 2018-01-30:
I have the same problem.
In the new version avrdude, the switch "I" is no longer available.

This doesn't work anymore:
avrdude -p [Board] -c arduino -P [USB Port from Octoprint] -b 57600 -D -U flash:w:[file you uploaded earlier]:I

This works:
avrdude -p m1284p -c arduino -P [USB Port from Octoprint] -b 57600 -D -U flash:w:[file you uploaded earlier]

This must be changed in the program code. :-)

benlye, 2018-01-30:
Well that's annoying!  Which version do you have, and do you have a link to any updated docs?

grandmastermp, 2018-01-30:
Here is my installed version:
avrdude version 6.1, URL: <http://savannah.nongnu.org/projects/avrdude/>

This version was automatically installed through the library.

benlye, 2018-01-30:
I have v6.3, which is the current one, and it works fine with the `:i` at the end of the command.  Could you try updating?

grandmastermp, 2018-01-30:
Interesting. Why is an old version installed over the library?
How do I get this 6.3 installed?

benlye, 2018-01-30:
The plugin doesn't self-install avrdude - you have to do it yourself.  I guess the one you have came with something else on your system.

Try this in a shell on your OctoPrint server:
```
sudo apt-get update
sudo apt-get install --only-upgrade avrdude
```

riisefrutti, 2018-01-30:
I just did a `sudo apt-get install avrdude` - it states I have the newest version.

Here is my log from FWUpdate:
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1679073/plugin_firmwareupdater_console.log)

grandmastermp, 2018-01-30:
Yes, the same:

pi@octopi:~ $ sudo apt-get install --only-upgrade avrdude
Reading package lists... Done
Building dependency tree       
Reading state information... Done
avrdude is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

benlye, 2018-01-30:
Did you do this first:
`sudo apt-get update`

riisefrutti, 2018-01-30:
Yes, I did.

grandmastermp, 2018-01-30:
Yes, I did that first.

benlye, 2018-01-30:
@riisefrutti, you're getting checksum errors when avrdude is reading the file back.  So the write works, but the read is a mess:
```
2018-01-30 11:56:18,605 > avrdude: input file /tmp/tmp3FANkG contains 170136 bytes
2018-01-30 11:56:18,606 > avrdude: reading on-chip flash data:
2018-01-30 11:56:18,607 > 
2018-01-30 11:56:18,607 > 
2018-01-30 11:56:21,275 > Reading | #####avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:21,276 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:21,277 > 
2018-01-30 11:56:22,443 > ####avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:22,444 > 
2018-01-30 11:56:23,573 > ##avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:23,575 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:23,575 > 
2018-01-30 11:56:25,002 > ###avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:25,003 > avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:25,004 > avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:25,004 > 
2018-01-30 11:56:25,920 > ###avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:25,921 > 
2018-01-30 11:56:27,100 > ##avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:27,102 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:27,103 > 
2018-01-30 11:56:28,020 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:28,021 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:28,022 > 
2018-01-30 11:56:29,039 > ###avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:29,044 > 
2018-01-30 11:56:32,211 > ########avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:32,213 > 
2018-01-30 11:56:33,528 > ###avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:33,529 > #avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:33,530 > 
2018-01-30 11:56:34,295 > ##avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:34,300 > 
2018-01-30 11:56:35,717 > ###avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:35,718 > 
2018-01-30 11:56:37,648 > #####avrdude: stk500v2_recv(): checksum error
2018-01-30 11:56:37,649 > # | 100% 19.00s
2018-01-30 11:56:37,650 > 
2018-01-30 11:56:37,650 > avrdude: verifying ...
2018-01-30 11:56:37,651 > avrdude: verification error, first mismatch at byte 0x4a17
2018-01-30 11:56:37,652 >          0xe0 != 0x94
2018-01-30 11:56:37,653 > avrdude: verification error; content mismatch
2018-01-30 11:56:37,653 > 
2018-01-30 11:56:37,654 > avrdude done.  Thank you.
```

I think you should work out how to get the latest avrdude installed on your system.

@grandmastermp, does your log look similar?

riisefrutti, 2018-01-31:
Well, after compiling avrdude 6.3 the error still persist.  There is still a communication error. Write  works, read does not.

[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1680052/plugin_firmwareupdater_console.log)

benlye, 2018-01-31:
> Well, after compiling avrdude 6.3 the error still persist. There is still a communication error. Write works, read does not.

What happens if you copy the compiled hex file somewhere locally on your Pi and try to run the avrdude command manually?  Do you get the same errors?

What happens if you add `-b 57600` to try a slower baud rate?

Presumably you've been able to flash this board without errors by other means?

grandmastermp, 2018-01-31:
Hi,
I've changed the debian-libraries from Jessie to Stretch and updated again. Now I have the version 6.3.
I'll test tonight. :-)

benlye, 2018-01-31:
@grandmastermp did you check your log?  Are you seeing the same verification error, or something else?

Google suggests that write verification errors can sometimes be fixed by re-flashing the board's bootloader.  You would need a USBasp or another external programmer to do that.

In the meantime I'm going to add detection for a verification error so that you don't get 'Unknown error' if it is the verification that fails.  I can also add an option to disable write verification, but that feels like a bad idea.  If it was my board I'd want to fix the problem.

grandmastermp, 2018-01-31:
@benlye 
I have not checked yet. But with version 6.1, I can update the firmware manually via command line.

benlye, 2018-01-31:
> I have not checked yet. But with version 6.1, I can update the firmware manually via command line.

OK.  It would be good to see the plugin log to see if you have the same problem or something else.

grandmastermp, 2018-01-31:
If I have problems with write verification errors, then the manual way should not work. 
Or what do you mean?

benlye, 2018-01-31:
You might be getting the 'unknown error' for a different reason'.  I'd like to know if it is for write verification or not.

If you are getting write verification errors from the plugin, it would be very interesting to know if the manual command works or not.  If the manual avrdude command doesn't give errors I'll need to know what it's doing differently to the command the plugin uses.

The plugin doesn't do anything special - it just runs the avrdude command, so the results should be the same...

riisefrutti, 2018-01-31:
I ran this today, successfully,:
`i@tarantula:~ $ /usr/bin/avrdude -C/etc/avrdude.conf -v -v -v -v \
> -patmega2560 -cwiring -P/dev/ttyUSB0 -b115200 -D \
> -Uflash:w:~/FW/Marlin.hex:i

avrdude: Version 6.1, compiled on May 29 2015 at 01:38:36
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
avrdude: wiring_open(): releasing DTR/RTS
avrdude: wiring_open(): asserting DTR/RTS
avrdude: Send: . [1b] . [01] . [00] . [01] . [0e] . [01] . [14] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
[... 736 lines cut ...]
avrdude: Recv: . [00] 
avrdude: Recv: . [07] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [1d] 
avrdude: Recv: . [00] 
avrdude: Recv: . [00] 
avrdude: Recv: P [50] 
avrdude: Recv: . [00] 
avrdude: Recv: . [ff] 
avrdude: Recv: . [00] 
avrdude: Recv: . [bb] 
avrdude: safemode read 3, efuse value: ff
avrdude: safemode: efuse reads as FF
avrdude: safemode: Fuses OK (E:FF, H:D8, L:FF)
avrdude: Send: . [1b] . [1c] . [00] . [03] . [0e] . [11] . [01] . [01] . [1b] 
avrdude: Recv: . [1b] 
avrdude: Recv: . [1c] 
avrdude: Recv: . [00] 
avrdude: Recv: . [02] 
avrdude: Recv: . [0e] 
avrdude: Recv: . [11] 
avrdude: Recv: . [00] 
avrdude: Recv: . [1a] 

avrdude done.  Thank you.`

riisefrutti, 2018-01-31:
6.3 log:
` /home/pi/avrdude-6.3/avrdude -v -p m2560 -c wiring -P /dev/ttyUSB0 -U flash:w:/home/pi/FW/marlin.hex:i -D

avrdude: Version 6.3, compiled on Jan 30 2018 at 21:40:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/usr/local/etc/avrdude.conf"
         User configuration file is "/home/pi/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : wiring
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as FD
avrdude: reading input file "/home/pi/FW/marlin.hex"
avrdude: writing flash (170268 bytes):

Writing | ################################################## | 100% 24.66s

avrdude: 170268 bytes of flash written
avrdude: verifying flash memory against /home/pi/FW/marlin.hex:
avrdude: load data flash data from input file /home/pi/FW/marlin.hex:
avrdude: input file /home/pi/FW/marlin.hex contains 170268 bytes
avrdude: reading on-chip flash data:

Reading | ########                                           | 16% 3.05savrdude: stk500v2_recv(): checksum error
Reading | #########                                          | 18% 3.42savrdude: stk500v2_recv(): checksum error
[... 16 lines cut ...]
Reading | ######################################             | 76% 19.50savrdude: stk500v2_recv(): checksum error
Reading | ########################################           | 80% 20.25savrdude: stk500v2_recv(): checksum error
Reading | ##########################################         | 83% 20.82savrdude: stk500v2_recv(): checksum error
Reading | ###########################################        | 85% 21.22savrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_getsync(): timeout communicating with programmer
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
a`

benlye, 2018-01-31:
@riisefrutti so you get checksum errors when you try to flash from the Pi command line as well.

A quick Google search suggests that re-flashing the board's bootloader may help.

riisefrutti, 2018-01-31:
The just reinstalled PI bootloader. Will try to Flash bootloader to the tarantula board.

Sendt fra min iPhone

grandmastermp, 2018-01-31:
Still the same error after the upgrade. :-(

Here the error message with switch "I" on the end:

avrdude -p m1284p -c arduino -P /dev/ttyUSB0 -b 57600 -D -U flash:w:Marlin.ino.sanguino.hex:I
avrdude: invalid file format 'I' in update specifier
avrdude: error parsing update operation 'flash:w:Marlin.ino.sanguino.hex:I'

And here the flash process without "I":

avrdude -p m1284p -c arduino -P /dev/ttyUSB0 -b 57600 -D -U flash:w:Marlin.ino.sanguino.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file "Marlin.ino.sanguino.hex"
avrdude: input file Marlin.ino.sanguino.hex auto detected as Intel Hex
avrdude: writing flash (112158 bytes):

Writing | ################################################## | 100% 27.00s

avrdude: 112158 bytes of flash written
avrdude: verifying flash memory against Marlin.ino.sanguino.hex:
avrdude: load data flash data from input file Marlin.ino.sanguino.hex:
avrdude: input file Marlin.ino.sanguino.hex auto detected as Intel Hex
avrdude: input file Marlin.ino.sanguino.hex contains 112158 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 22.51s

avrdude: verifying ...
avrdude: 112158 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

benlye, 2018-01-31:
It should be a lower case i - `:i`, not `:I`.

grandmastermp, 2018-01-31:
Ok.
On the Internet I have found instructions, there is "I" specified.

grandmastermp, 2018-01-31:
Yes, it works:

avrdude -p m1284p -c arduino -P /dev/ttyUSB0 -b 57600 -D -U flash:w:Marlin.ino.sanguino.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: reading input file "Marlin.ino.sanguino.hex"
avrdude: writing flash (112158 bytes):

Writing | ################################################## | 100% 27.00s

avrdude: 112158 bytes of flash written
avrdude: verifying flash memory against Marlin.ino.sanguino.hex:
avrdude: load data flash data from input file Marlin.ino.sanguino.hex:
avrdude: input file Marlin.ino.sanguino.hex contains 112158 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 22.51s

avrdude: verifying ...
avrdude: 112158 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

grandmastermp, 2018-01-31:
But why doesn't work with your tool?

benlye, 2018-01-31:
> But why doesn't work with your tool?

I don't know, that's why I've asked you to attach the log...

benlye, 2018-01-31:
@grandmastermp I think I see your problem (but I still want to see your log).

In your working example above you are setting the baud rate with `-b 57600`.  The default baud rate for an atmega1284p is 115200, and that is probably what is in the command the plugin is using.  The log would confirm it.

I'll add a configurable setting to allow overriding the default avrdude baud rate with a custom one.  In the meantime, if you want to, you could edit this file:
`~/OctoPrint/venv/lib/python2.7/site-packages/octoprint_firmwareupdater/__init__.py`, 

find this line:
` avrdude_command = [avrdude_path, "-v", "-p", avrdude_avrmcu, "-c", avrdude_programmer, "-P", printer_port, "-U", "flash:w:" + firmware + ":i", "-D"]`

and change it to this:
` avrdude_command = [avrdude_path, "-v", "-p", avrdude_avrmcu, "-c", avrdude_programmer, "-P", printer_port, "-U", "flash:w:" + firmware + ":i", "-D -b 57600"]`

grandmastermp, 2018-02-01:
@benlye Under the directory "OctoPrint" I can not find "venv".

benlye, 2018-02-01:
Oh well.  Can you find the log?  I have a new version for you to test, but until I see the log I won't know if it will fix your problem...

grandmastermp, 2018-02-01:
Here's my log:

2018-01-31 17:58:36,764 /usr/bin/avrdude -v -p m1284p -c arduino -P /dev/ttyUSB0 -U flash:w:/tmp/tmpkrhPq6:i -D
2018-01-31 17:58:37,422 > 
2018-01-31 17:58:37,423 > avrdude: Version 6.3
2018-01-31 17:58:37,424 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2018-01-31 17:58:37,425 >          Copyright (c) 2007-2014 Joerg Wunsch
2018-01-31 17:58:37,426 > 
2018-01-31 17:58:37,427 >          System wide configuration file is "/etc/avrdude.conf"
2018-01-31 17:58:37,427 >          User configuration file is "/home/pi/.avrduderc"
2018-01-31 17:58:37,428 >          User configuration file does not exist or is not a regular file, skipping
2018-01-31 17:58:37,429 > 
2018-01-31 17:58:37,430 >          Using Port                    : /dev/ttyUSB0
2018-01-31 17:58:37,431 >          Using Programmer              : arduino
2018-01-31 17:58:37,431 > 
2018-01-31 17:58:40,115 > avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x73
2018-01-31 17:58:40,116 > avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x74
2018-01-31 17:58:40,117 > avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x61
2018-01-31 17:58:40,118 > avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
2018-01-31 17:58:40,118 > avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x74
2018-01-31 17:58:40,119 > avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x0a
2018-01-31 17:58:40,120 > avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x65
2018-01-31 17:58:40,120 > avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x63
2018-01-31 17:58:40,121 > avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x68
2018-01-31 17:58:40,122 > avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6f
2018-01-31 17:58:40,123 > 
2018-01-31 17:58:40,123 > avrdude done.  Thank you.

grandmastermp, 2018-02-01:
"not in sync" = serial speed.

benlye, 2018-02-01:
Use the Plugin Manager to install this version:
`https://github.com/benlye/OctoPrint-FirmwareUpdater/archive/devel.zip`

I've added 'Advanced Settings' where you can change the avrdude baud rate and also disable write verification.  I've done a fair bit of testing already.  If it works for you I'll release it.

[image]

grandmastermp, 2018-02-01:
Yes, it works. Thank you very much. :-)

2018-02-01 18:55:56,754 /usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -b 57600 -U flash:w:/tmp/tmpHDtyjF:i
2018-02-01 18:55:57,401 > 
2018-02-01 18:55:57,403 > avrdude: Version 6.3
2018-02-01 18:55:57,403 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2018-02-01 18:55:57,404 >          Copyright (c) 2007-2014 Joerg Wunsch
2018-02-01 18:55:57,405 > 
2018-02-01 18:55:57,405 >          System wide configuration file is "/etc/avrdude.conf"
2018-02-01 18:55:57,406 >          User configuration file is "/home/pi/.avrduderc"
2018-02-01 18:55:57,406 >          User configuration file does not exist or is not a regular file, skipping
2018-02-01 18:55:57,407 > 
2018-02-01 18:55:57,407 >          Using Port                    : /dev/ttyUSB0
2018-02-01 18:55:57,408 >          Using Programmer              : arduino
2018-02-01 18:55:57,409 >          Overriding Baud Rate          : 57600
2018-02-01 18:55:57,409 > 
2018-02-01 18:55:58,559 >          AVR Part                      : ATmega1284P
2018-02-01 18:55:58,560 >          Chip Erase delay              : 55000 us
2018-02-01 18:55:58,561 >          PAGEL                         : PD7
2018-02-01 18:55:58,561 >          BS2                           : PA0
2018-02-01 18:55:58,562 >          RESET disposition             : dedicated
2018-02-01 18:55:58,563 >          RETRY pulse                   : SCK
2018-02-01 18:55:58,563 >          serial program mode           : yes
2018-02-01 18:55:58,564 >          parallel program mode         : yes
2018-02-01 18:55:58,565 >          Timeout                       : 200
2018-02-01 18:55:58,565 >          StabDelay                     : 100
2018-02-01 18:55:58,566 >          CmdexeDelay                   : 25
2018-02-01 18:55:58,567 >          SyncLoops                     : 32
2018-02-01 18:55:58,568 >          ByteDelay                     : 0
2018-02-01 18:55:58,568 >          PollIndex                     : 3
2018-02-01 18:55:58,569 >          PollValue                     : 0x53
2018-02-01 18:55:58,570 >          Memory Detail                 :
2018-02-01 18:55:58,570 > 
2018-02-01 18:55:58,571 >                                   Block Poll               Page                       Polled
2018-02-01 18:55:58,572 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2018-02-01 18:55:58,573 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2018-02-01 18:55:58,574 >            eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
2018-02-01 18:55:58,575 >            flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
2018-02-01 18:55:58,576 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2018-02-01 18:55:58,582 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2018-02-01 18:55:58,583 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2018-02-01 18:55:58,584 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2018-02-01 18:55:58,585 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2018-02-01 18:55:58,586 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2018-02-01 18:55:58,588 > 
2018-02-01 18:55:58,589 >          Programmer Type : Arduino
2018-02-01 18:55:58,591 >          Description     : Arduino
2018-02-01 18:55:58,592 >          Hardware Version: 2
2018-02-01 18:55:58,592 >          Firmware Version: 1.16
2018-02-01 18:55:58,593 >          Vtarget         : 0.0 V
2018-02-01 18:55:58,594 >          Varef           : 0.0 V
2018-02-01 18:55:58,595 >          Oscillator      : Off
2018-02-01 18:55:58,595 >          SCK period      : 0.1 us
2018-02-01 18:55:58,596 > 
2018-02-01 18:55:58,597 > avrdude: AVR device initialized and ready to accept instructions
2018-02-01 18:55:58,598 > avrdude: Device signature = 0x1e9705 (probably m1284p)
2018-02-01 18:55:58,600 > avrdude: reading input file "/tmp/tmpHDtyjF"
2018-02-01 18:55:58,601 > avrdude: writing flash (112158 bytes):
2018-02-01 18:55:58,602 > 
2018-02-01 18:56:25,626 > avrdude: 112158 bytes of flash written
2018-02-01 18:56:25,627 > avrdude: verifying flash memory against /tmp/tmpHDtyjF:
2018-02-01 18:56:25,628 > avrdude: load data flash data from input file /tmp/tmpHDtyjF:
2018-02-01 18:56:25,629 > avrdude: input file /tmp/tmpHDtyjF contains 112158 bytes
2018-02-01 18:56:25,629 > avrdude: reading on-chip flash data:
2018-02-01 18:56:25,630 > 
2018-02-01 18:56:48,162 > avrdude: verifying ...
2018-02-01 18:56:48,163 > avrdude: 112158 bytes of flash verified
2018-02-01 18:56:48,165 > 
2018-02-01 18:56:48,166 > avrdude done.  Thank you.
2018-02-01 18:56:48,167 > 
2018-02-01 18:56:48,168 > 
2018-02-01 18:56:48,170 Flashing successful.

riisefrutti, 2018-02-01:
On my side, no help with the new bootloader. Im going to try with another RPI, and a fresh install. Having a feeling there might be something with the PITFT jessie(adafruit) that prevents the reading. Normal upload of FW with arduino IDE works like a charm.

grandmastermp, 2018-02-02:
@riisefrutti Have you tried the command line via ssh? As I did?

riisefrutti, 2018-02-02:
As before SSH unsuccessful, only read error.

grandmastermp, 2018-02-02:
I think the problem is not the plugin.

riisefrutti, 2018-02-02:
@grandmastermp I totally agree. 

@benlye Thank you for the help. The last update you posted will get me through as of now, and Ill sort the rest out later.

benlye, 2018-02-02:
Cool. I'll merge the PR this morning.

grandmastermp, 2018-02-02:
Perfect. :-)

benlye, 2018-02-02:
Merged #31.

closed 2018-02-02 by benlye