OctoPrint/OctoPrint-FirmwareUpdater #33: Arduino DUE Support

Laudix64, 2018-02-02:
Hi.

Can you add RADDS and Arduino Due support?

Best regards

Salvatore

benlye, 2018-02-02:
Yes, probably.

Can you help me out by giving me an example avrdude command line?  One that the Arduino IDE generates, or one that you can use on the console?

benlye, 2018-02-02:
Looking into this the Due uses bossac rather than avrdude.  The plugin was architected to support different flashing tools but right now is only set up with avrdude.  Adding bossac as a flashing method won't be too hard, but I won't be able to test it as I don't have a Due to try to flash.

If you can give me an example command line and a few days I'll get something together for you to test.

Laudix64, 2018-02-03:
Hi.
Sorry for my english...
This is the command launched by the arduino IDE:
**C:\Users\xxxxxx\Documents\ArduinoData\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM9 -U false -e -w -v -b C:\Users\laudicin\AppData\Local\Temp\arduino_build_614472/MK4duo.ino.bin -R** 
I hope I have been helpful, I'm available to help you with the tests.

Best regards

Salvatore

benlye, 2018-02-03:
OK, I've had a go at adding the new method. I had to re-do the plugin's settings UI a bit.  Once configured the plug-in will run a command that looks like this:
`/usr/bin/bossac -i -d -p /dev/ttyACM0 -U false -e -w -v -b /tmp/tmpOGgXb9 -R`
Which looks like the one the IDE issues.

You can install the test version manually using the plug-in manager from here:
https://github.com/benlye/OctoPrint-FirmwareUpdater/archive/devel.zip

I'm reasonably confident it will work.  I'm interested to know if the progress bar in the UI updates correctly.  Whether it works or not, can you send me the octoprint.log file and the plugin_firmwareupdater_console.log file so I can confirm what success looks like?

benlye, 2018-02-04:
By the way, you need to install bossac on your OctoPrint server with the command line `sudo apt-get install bossa-cli` (assuming it's OctoPi or similar).

Laudix64, 2018-02-04:
Hi.

The plugin does not work, it seems that it does not find the serial port; I attach the logs you requested.

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1693026/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1693027/plugin_firmwareupdater_console.log)

Ciao

Salvatore

Laudix64, 2018-02-04:
[image]
[image]

benlye, 2018-02-04:
Which port is the OctoPrint server connected to, native or programming? Can you try the other one?

Laudix64, 2018-02-04:
Hi.

I use programming port, with MK4DUO firmware the native port does not work.

[image]

Bye

Salvatore

benlye, 2018-02-04:
OK, comparing to the Arduino IDE it looks like the programming port is the one bossac wants to use, so that should be OK.  I think the problem may be that  the due needs to be reset/erased before flashing starts:
https://playground.arduino.cc/Botloader/DueBootloaderExplained

It seems like the Arduino IDE installs a [variant of bossac](https://github.com/shumatech/BOSSA/tree/arduino) which is tweaked to do this automatically, and I assume this is not the same as the version installed by apt-get.

Are you willing to try pressing the 'erase' button on your Due before attempting to flash it with the plugin?  I don't have a Due and only know what little I've read about programming them.  I don't know what it will do to the Due or what you might need to do to get the firmware back onto it afterwards.  Presumably you could re-flash it from the Arduino IDE.

If that does work it seems like there are a couple of ways to trigger the reset/erase so I could incorporate that into the plug-in's flash routine.

benlye, 2018-02-04:
Actually, can you try flashing the firmware using the Native port? Even though OctoPrint cannot connect to it, the MK4Duo docs say to use the Native port for firmware uploads.

https://github.com/MKFirmware/MK4duo/blob/master/Documentation/Compilation.md

It could be that you need one port for OctoPrint and another port for firmware flashing.

Ben

benlye, 2018-02-04:
Doing more reading...

The arduino version of bossac was merged into the main branch a year or so ago, before version 1.7.0, so that's not an issue.  But... apt-get installed version 1.3a of bossac on my Raspberry Pi, so it looks like the new version needs to be built from source.
```
cd ~/
sudo apt-get install libwxgtk3.0-dev libreadline-dev 
wget https://github.com/shumatech/BOSSA/archive/master.zip
unzip master.zip
cd BOSSA-master
make
```

Then, the Arduino IDE has a flag (`use_1200bps_touch`) in the Due board configuration to make an initial connection to the board at 1200bps which triggers the reset and erase and preps the board for flashing.  I'll have to build that into the process.

I've ordered a cheap Due clone from eBay.  Once I get my hands on it I should be able to get this working :-)

Ben

Laudix64, 2018-02-05:
Hi Ben.

I tried to connect both USB ports, but with none it works
Bye

Salvatore

Laudix64, 2018-02-05:
Hi Ben.

Have you seen this link?
https://github.com/johngrantuk/piupdue

Bye.

Salvatore

benlye, 2018-02-05:
I think I've got a method to do the reset before trying to use bossac to flash the board.  As soon as the Due I've ordered arrives I'll be able to give it a go.  If that doesn't work I'll look at piupdue.

Thanks!

Ben

benlye, 2018-02-13:
I got my Due board and have this working.  It only works with specific versions of bossac - the one that comes with the Arduino IDE, or v1.7.0.  It does not work with 1.8.0 or the master, as they seem to have changed the options and removed some of the Arduino-specific goodies (such as the ability to reset the bard with the `-R` flag).  That said, it works reliably with v1.7.0 for me.

1. Download, compile, and install the working bossac tool:
```
cd ~/
sudo apt-get install libwxgtk3.0-dev libreadline-dev 
wget https://github.com/shumatech/BOSSA/archive/1.7.0.zip
unzip 1.7.0.zip
cd BOSSA-1.7.0
./arduino/make_package.sh
sudo cp ~/BOSSA-1.7.0/bin/bossac /usr/local/bin/
```

2. Install the devel version of the Firmware Updater plugin from here:
`https://github.com/benlye/OctoPrint-FirmwareUpdater/archive/devel.zip`

3. Configure the plugin for bossac and use the new path to bossac: `/usr/local/bin/bossac`

4. Try to flash!

Let me know how it goes.

Ben

benlye, 2018-02-20:
@Laudix64 did you have a chance to test?

Laudix64, 2018-02-20:
Hi Ben.

I read your message only today, my printer are in mantenance.
I try as soon as possible

Bye

Salvatore

Hazel273, 2018-02-26:
Tested the last due flashing method and im happy to report it works, great job!

-Hazel

benlye, 2018-02-27:
Thanks, @Hazel273.  I have a few doc updates to finish then I'll push the new release out.

benlye, 2018-03-13:
I've released v1.0.0 with Due support!

closed 2018-03-13 by benlye