OctoPrint/OctoPrint-FirmwareUpdater #34: Flash failing to ramps 1.4

cyris69, 2018-02-04:
I was never able to get this plugin when it first came out with stock creality board but was able to do it manually in terminal and though I'd give it a go at its current updated version. I'm compiling a hex I know works as it's flashable over usb from PC. I swapped to ramps due to crealities horrible hardware and decaying performance (which ramps has almost revived).

Log is reporting this error:

```
2018-02-04 04:43:20,692 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2018-02-04 04:43:20,705 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2018-02-04 04:43:20,820 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-02-04 04:43:20,823 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin TH3D U1.R1"
2018-02-04 04:43:34,763 - octoprint.server.preemptive_cache - INFO - ... done in 31.02s
2018-02-04 04:43:34,764 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=de', 'path': '/', 'base_url': 'http://192.168.1.16/'}
2018-02-04 04:43:34,854 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.111
2018-02-04 04:43:36,636 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.111
2018-02-04 04:43:46,029 - octoprint.server.preemptive_cache - INFO - ... done in 11.27s
2018-02-04 04:44:21,695 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2018-02-04 04:44:21,703 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2018-02-04 04:44:21,741 - octoprint.plugins.firmwareupdater - INFO - Running u'/usr/bin/avrdude -v -q -p m2560 -c arduino -P /dev/ttyUSB0 -D -b 115200 -U flash:w:/tmp/tmpmRzLeI:i' in /usr/bin
2018-02-04 04:45:13,757 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude returned code 1.
2018-02-04 04:45:13,763 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'name': 'CR-10', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 300, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 300.0, 'width': 300.0, 'custom_box': False, 'height': 400.0}, 'model': 'Creality', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.5, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2018-02-04 04:45:13,864 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-02-04 04:45:14,941 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-02-04 04:45:23,334 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2018-02-04 04:45:23,350 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2018-02-04 04:45:23,361 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2018-02-04 04:45:23,407 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin TH3D U1.R1"
```

benlye, 2018-02-04:
The 'not in sync: resp=0x00' error can be caused by many things.  

Sometimes it means the baud rate is wrong.  Have you tried flashing without specifying a baud rate for avrdude?  Note that the baud rate needed for flashing will not necessarily be the same as the baud rate used by Octoprint to talk to the printer - it needs to match the rate of whatever bootloader you have installed.

It could also be an issue with the USB connection between the OctoPrint server and the board.

Can you give an example of a working command when you flash from PC?  Does that same command work if you use it from the OctoPrint command line?

I flash my mega2560 RAMPS board using the plugin nearly daily, so I know it does work.

benlye, 2018-02-04:
Disregard my previous response.

Change the programmer type from `arduino` to `wiring`.

closed 2018-02-07 by benlye

cyris69, 2018-02-15:
That worked, sorry, didn't get an email update about response.