OctoPrint/OctoPrint-FirmwareUpdater #39: [Enhancement] Button(s) for "favorite" File(s) or URL(s)

b-morgan, 2018-03-26:
I have multiple extruders (Single, Dual, Flexible) that each have a unique firmware. Would it be possible to add some buttons that could be pre-configured with often used file or URL locations?

hamster65, 2018-03-30:
I thought it would be handy if the plugin remembered the last used file so that you do have to browse for it. For me, the firmware binary is always in the same folder. Having buttons with different files would cover this. Good idea!

cmidgley, 2019-01-26:
I also would like to up-vote this request - in my case it's the Lulzbot Taz with multiple extruders, and it would be awesome if the firmware could be uploaded to the Pi, and we could select which firmware we wanted loaded.  Even more amazing would be a way to inject a hint in the g-code as to which firmware is required (aka, which print head we sliced for), and we could detect if the correct firmware is available and refuse to print until the upload is complete.

benlye, 2020-11-28:
Unfortunately this isn't possible due to the way that the browser interacts with the filesystem (for security reasons).  You have to use the 'Browse' button to open the file and read the data from it.  Pretty much the same reason why #141 and #157 aren't possible either.

closed 2020-11-28 by benlye