OctoPrint/OctoPrint-FirmwareUpdater #42: Upgrading firmware results in EEPROM CRC mismatch

b-morgan, 2018-04-08:
Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: EEPROM CRC mismatch - (stored) 31524 != 14490 (calculated)!

OctoPrint 1.3.7rc4 running on OctoPi 0.14.0, Firmware updater 1.0.0, LulzBot TAZ 6, Marlin 1.1.8.22

I believe this is caused by a recent "update" to Marlin 1.1.8 and it is resolved by doing a "Save Settings" from the LCD on the printer. It should also be resolved by sending an "M500" to the printer.

I checked "Enable post-flash gcode" and in "Post-flash gcode" I put "M500" but this fails because OctoPrint never gets a chance to send the post-flash gcode.

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1887895/octoprint.log)
[serial.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/1887896/serial.log)

b-morgan, 2018-04-09:
From @foosel [https://github.com/foosel/OctoPrint/issues/2548#issuecomment-379664437](url)
> Not a bug in OctoPrint, if the firmware is sending a "Error" message and you have it configured to disconnect on those, that's completely valid behaviour (and has always been present). Sounds more like a bug in the firmware to be honest that it sends an "Error" here in the first place. The FirmwareUpdater plugin - whose maintenance I btw handed off, no time - could work around this (in 1.3.7) by capturing the problematic error message and ignoring it.
[http://docs.octoprint.org/en/maintenance/plugins/hooks.html#octoprint-comm-protocol-gcode-error](url)

benlye, 2018-04-11:
I guess I don't understand what's going on - I've never seen this issue.  

Based on your description, my understanding was that the firmware was getting updated, then you got an EEPROM CRC error on bootup (causing OctoPrint to fail to connect), and the fix was to use `M500` to save the settings (which will now be defaulted) to the EEPROM.  

Can you not just save the defaulted settings from the LCD then reboot?

benlye, 2018-04-11:
Hmm, are you sure you're not just running into a known bug, which was subsequently fixed:
https://github.com/MarlinFirmware/Marlin/pull/8283

Are you able to try the latest Marlin bugfix-1.1.x branch to see if you keep getting errors?

b-morgan, 2018-04-11:
> Can you not just save the defaulted settings from the LCD then reboot?

Yes, I can do that but the printer and the Raspberry Pi are in a different room than the desktop where I slice and control OctoPrint. 

I have to change firmware every time I change toolheads (from single to dual and vice versa). I change the toolhead on the printer, walk back to my desktop, update the firmware, start preheating the extruder(s), slice the part I want to print, upload it, start the print, and then monitor progress for a few layers either remotely via the Pi camera or physically by walking back to the printer. I'm just trying to save the extra trip to the printer and back just to play with the LCD.

> Hmm, are you sure you're not just running into a known bug, which was subsequently fixed

I checked the Marlin sources that LulzBot is using and that fix is included.

foosel, 2018-04-11:
What about only disabling that error when you know you just flashed before hand? E.g. set a var to `True` on successful flashing, check for that and reset it after the next successful connect?

To be honest though, the whole behaviour sounds more like a firmware bug than intentional.

As a side note, you can also tell OctoPrint to not disconnect on any unhandled `Error` messages it receives, that would solve the problem at least until some kind of agreement is reached.

benlye, 2018-04-11:
> To be honest though, the whole behaviour sounds more like a firmware bug than intentional.

I agree.  I'm going to try and reproduce this later with a spare Mega2560 board I have and Marlin code from around the 1.1.8 release. If I can, I'll look at adding an option to handle the CRC error.

@b-morgan are you flashing a pre-compiled firmware image that you've downloaded, or one you are compiling yourself from source?  Can you point me at where it comes from?

benlye, 2018-04-12:
I've just flashed my Mega2560 several times with both Marlin 1.1.7 and 1.1.8 and have not been able to get it to give me an EEPROM CRC error even once, even going between versions.

@b-morgan I'm thinking that whatever is going on your system isn't normal, and is most likely a bug in the firmware you are flashing (or, possibly, but much less likely, a hardware problem).

I don't think it's the Firmware Update plugin's job to detect and suppress these errors after a flash.  I wouldn't that on my printer.

Also, given that you can globally tell OctoPrint to ignore the error and connect anyway, I think you have a workaround until the underlying firmware issue is resolved in your Marlin fork.

I can see that there could be use for a plugin which detects errors from the printer and lets users configure responses (including 'ignore'), like the [Action Commands Plugin](https://plugins.octoprint.org/plugins/actioncommands/) handles `//action:` output.  It wouldn't be too hard to create a new plugin code to do it.

b-morgan, 2018-05-26:
Just to provide some additional information for anyone stumbling across this issue. If the old firmware has EXTRUDERS 1 and the new firmware has EXTRUDERS 2 (or vice versa) then this "bug" is triggered.

Alissonverd, 2019-03-05:
I just went through the same mistake.
Changing BLTouch or Servo probe to Solenoid probe will trigger this bug.

So, to solve it.. Clear your EEPROM with this code: https://www.arduino.cc/en/Tutorial/EEPROMClear

=D
Best Regards

Fips11, 2019-03-23:
My solution without flashing anything was.
- Go to the octoprint settings. 
- In the settings under Serial connection -> Behavior -> Cancel current print jobs and then execute M500 in the terminal. 
After that Marlin was usable normally.

(Bitte verzeiht mir eigenartige Ausdrucksweisen der das ist nur eine deepl Übersetzung.)