OctoPrint/OctoPrint-FirmwareUpdater #43: Enable post-flash gcode sends a command when the printer connects without a firmware update

b-morgan, 2018-04-11:
I have changed OctoPrint's Serial Connection Behavior from "disconnect on error" to "cancel print but stay connected". This allows the post-flash gcode to be executed after a firmware update (which works).

The problem is that when OctoPrint connects to the printer and there was not a firmware update, an unknown command is sent.

`2018-04-11 07:34:51,840 - Changing monitoring state from "Connecting" to "Operational"
2018-04-11 07:34:51,858 - Send: N0 M110 N0*125
2018-04-11 07:34:51,860 - Recv: echo:SD card ok
2018-04-11 07:34:51,866 - Recv: echo:Unknown command: ""
2018-04-11 07:34:51,869 - Recv: ok P15 B9
2018-04-11 07:34:51,871 - Send: N1 M115*39
2018-04-11 07:34:51,898 - Recv: FIRMWARE_NAME:Marlin  FIRMWARE_VERSION:1.1.8.22 EXTRUDER_TYPE:DualExtruder v3 SOURCE_CODE_URL:https://code.alephobjects.com/diffusion/MARLIN PROTOCOL_VERSION:1.0 MACHINE_TYPE:LulzBot TAZ 6 EXTRUDER_COUNT:2 UUID:845f003c-aebd-4e53-a6b9-7d0984fde609
`

benlye, 2018-04-11:
Are you sure that is coming from the FirmwareUpdater Plugin?  Does it happen if you disable or remove the plugin?  What does the octoprint.log file show?  (Look for a line like `octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event` and see what follows it - the plugin logs that stuff pretty verbosely.)

b-morgan, 2018-04-11:
I assumed it was Firmware Updater but I was wrong! OctoPrint in safe mode still has the error.

[https://github.com/foosel/OctoPrint/issues/2569](url)

benlye, 2018-04-11:
So we can close this issue?

b-morgan, 2018-04-11:
Yes

closed 2018-04-11 by b-morgan