OctoPrint/OctoPrint-FirmwareUpdater #44: Install AVRDude from interface

edwinm, 2018-04-27:
Many people are not comfortable with the terminal and command line.

Can't you make an "Install AVRDude" button?

jhgreen215, 2018-06-13:
I'm not sure if this warrants its own issue, but I am having trouble installing AVRDude on a fresh install of OctoPi, even after `apt-get update && apt-get upgrade`

Maybe this feature would help?

benlye, 2019-05-23:
I don't think I want to do this.  Sure, it would probably be easy enough on a Pi, but not everyone uses that platform, and the installation can vary by distribution.  Also, if the installation has issues when you try to do it manually on a freshly-installed OctoPi, running the same commands by a button click is unlikely to be successful.

closed 2019-05-23 by benlye