OctoPrint/OctoPrint-FirmwareUpdater #46: Flashing failed Avrdude says: 'not in sync: resp=0x00'

Dr-Gandalf, 2018-05-07:
I have my printer working and I configured the plug in as specified in the interfase but the,  I keep getting the message 

Flashing failed
Avrdude says: 'not in sync: resp=0x00'

is there a way to get the command line that plugin is using?

[image]

the card is a ultimachine Einsy Rambo 1.1

benlye, 2018-05-08:
Yes, either the plug-in's own logfile, or the main Octoprint logfile will show you the full command being run.  Both logs are found in Settings -> Logging.

In the Octoprint log it will look like this:
```
2018-05-03 17:55:59,113 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2018-05-03 17:55:59,116 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2018-05-03 17:55:59,142 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2018-05-03 17:55:59,156 - octoprint.plugins.firmwareupdater - INFO - Running u'/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyACM0 -D -U flash:w:/tmp/tmp5BX7_g:i' in /usr/bin
```

'not in sync' often means that the baud rate is not correct.  You might want to try overriding the default baud rate and setting it to 115200.

closed 2018-05-10 by benlye

Kab14523, 2018-08-03:
As this would be the starting point for everyone who sees the 'not in sync' error from avrdude to look for solutions: Try to set the baudrate to 115200 in the advanced settings of FirmwareUpdater. If this doesn't work than lower the baudrate to 57600 - this did the trick for me (Anet A8, octoprint on RPi3, marlin 1.1.9).