OctoPrint/OctoPrint-FirmwareUpdater #57: Anet flash generating an error

banichow, 2018-09-09:
Sorry to bother, I am trying to use the plugin and it is generating:

Avrdude says: 'not in sync: resp=0x73

I have the correct port and the software is installed. It is an Anet 1.0 board. Works good from the PC. I looked on the Octoprint discourse site and did not find much. Below is the screen shots.

Thanks 

[image]
[image]

benlye, 2018-09-10:
The plug-in definitely works with the Anet board - I've flashed mine many times.  Try changing baud rate setting to 57600, 115200, or even 2500000.

banichow, 2018-09-10:
Thanks for the information, I will try again. I have it set to 115200 and this works for standard gcode. Does OctoPrint have to be "Connected" to the printer?

benlye, 2018-09-10:
No, OctoPrint doesn't need to be connected - the plugin will just disconnect if it is.

zak062, 2018-09-17:
I was having the same issue as Banichow with a Alunar M505 (anet a8 clone). I changed the baud rate to 57600 and was able to flash the firmware. Thanks you benlye!

closed 2018-09-17 by benlye