OctoPrint/OctoPrint-FirmwareUpdater #58: "Error: EEPROM datasize error." after flashing

brandstaetter, 2018-09-17:
After flashing (Marlin bugfix-2.0.x branch) on my Ender-4 I get "Offline (Error: EEPROM datasize error.)" Is this a firmware bug or something with Octoprint/FirmwareUpdater?

I don't like having to physically go to the printer to "Initialize EEPROM" after flashing remotely...

benlye, 2018-09-17:
A few people have asked this question.  It's not a problem with the uploader, but is a behaviour you will see if any config changes have modified the amount of EEPROM storage that Marlin requires.  It could be that you have made a change to your settings, or some new feature has been enabled by default and now more storage is needed.

See this Marlin issues:
https://github.com/MarlinFirmware/Marlin/issues/11699

Or, it could be a bug in Marlin - see this issue for a way to figure out what's going on.
https://github.com/MarlinFirmware/Marlin/issues/11757

One mitigation is to set `M502;M500` as post-flash commands.

brandstaetter, 2018-09-18:
Ah, there's an option in OctoPrint not to disconnect on Error - that's going to help! Also thanks for the other info.

closed 2018-09-18 by brandstaetter