OctoPrint/OctoPrint-FirmwareUpdater #61: No longer works with Octoprint Version 1.3.10rc1

GhostlyCrowd, 2018-11-08:
There have been some changes made that affect plugins on this version. Your plugin is one of them.

FanDjango, 2018-11-08:
And here is the error message as seen in the log:

2018-11-08 17:27:19,577 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unexpected error.
Traceback (most recent call last):
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_firmwareupdater/__init__.py", line 282, in _flash_avrdude
    p = sarge.run(avrdude_command, cwd=working_dir, async=True, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
TypeError: __init__() got an unexpected keyword argument 'async'

GhostlyCrowd, 2018-11-08:
Thanks @FanDjango  I couldnt get mine to throw a log out for some stupid reason

benlye, 2018-11-09:
Seems like that should be easy enough to fix, but I need to be careful not to break compatibility with older versions.  I'll need to do some testing.

In the meantime, would you be willing to do some testing?  It looks like the fix may be a one-word change.

FanDjango, 2018-11-09:
Of course, I'll test any changes right away.

FanDjango, 2018-11-09:
Meanwhile, to get this temporarily working again, I have changed "async" to "async_" in the two place where it occurs in __init__.py.
It now works again, I updated the firmware of my printer successfully but of course that change is not backwards compatible with people not yet on 1.3.10rc1

FanDjango, 2018-11-13:
@jneiliii: Theoretically, the sarge version is the relevant one. But I suppose checking the octoprint version is safe enough, hopefully. Both of them should be "in sync".

Sarge changelog:

**0.1.5**
Released: 2018-06-18
Fixed #37: Instead of an OSError with a “no such file or directory” message, a ValueError is raised with a more informative “Command not found” message.
**Fixed #38: Replaced async keyword argument with async_, as async has become a keyword in Python 3.7.**
Fixed #39: Updated tutorial example on progress monitoring.

TylonHH, 2018-11-26:
when will the pull request released?

benlye, 2018-11-27:
It looks like 1.3.10RC3 should solve the problem.
https://github.com/foosel/OctoPrint/issues/2889#issuecomment-441052392

I'm going to hold off making any changes until that has been tested.  If I do make changes it seems like it would be more sensible to leverage `octoprint.util.commandline.CommandlineCaller` instead of calling Sarge directly.

benlye, 2018-11-28:
RC3 is out and the plugin is working again now.

closed 2018-11-28 by benlye