OctoPrint/OctoPrint-FirmwareUpdater #63: Has anyone tried this with ANET A8?

masterchop, 2018-11-12:
I wonder if anyone has try this plug in with anet a8, i am getting the thermal warning but i havent seen anyone done this with it.

benlye, 2018-11-27:
Yes, the plugin works with the Anet A8 motherboard.

closed 2018-11-27 by benlye

masterchop, 2018-11-28:
Is there any guide? oh the steps to follow for that model?
I know the steps to do it manually from PC are online but this plugin looks like is going to be easier but there isnt any steps on how to use it. Specially for the anet a8.

benlye, 2018-11-28:
I can't really give specific guidance for every printer out there, but I do have an A8.  You need to configure the plugin for an Atmega1284p, using arduino as the programmer.

Depending on the bootloader on your board you might need to change the baud rate in advanced settings.  It should work with either 57600 or 115200.

[image]