OctoPrint/OctoPrint-FirmwareUpdater #65: avrdude path test doesn't work

WizardPC, 2018-11-27:
First at all, thanks for your job !

It's the first time i try this plugin and i made a typo writing `usr/bin/avrdude` instead of `/usr/bin/avrdude`.

But in both cases the "Test" say the path is valid... but from terminal (putty) i got `-bash: usr/bin/avrdude: No such file or directory`

It's an issue or i misunderstand something ?

benlye, 2018-11-28:
You're right - that's weird, I don't know why the path test succeeds without the leading slash.

WizardPC, 2018-11-28:
Moreover, with my typo i can start the firmware update but nothing append. And it get stuck on "Starting...". No errors in the log.

Madhaus01, 2018-12-07:
I had the same mistake
With me it works like that:
//usr/bin/avrdude
I do not know why either

Stady234, 2018-12-10:
Just tested with this path on 3 different pi (all my printers) and the path now validates!!

benlye, 2018-12-13:
v1.1.0 released which improves the path tests.

closed 2018-12-13 by benlye