OctoPrint/OctoPrint-FirmwareUpdater #66: Starting flash... over and over again. CH340

Zicmu69, 2018-12-01:


benlye, 2018-12-01:
Can you post the firmware updater and OctoPrint log files?

Zicmu69, 2018-12-09:
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/2660957/plugin_firmwareupdater_console.log)

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/2660963/octoprint.log)

benlye, 2018-12-11:
Check the path to avrdude - do you have it set to `/usr/bin/avrdude`?

If that doesn't work please retrieve and send the logs again - the OctoPrint log you've attached does not cover the same time period as the plugin log, and does not seem to contain an attempt to update the firmware.

Zicmu69, 2018-12-12:
Patch to avrdude is valid. Test prove it.
Here are latest logs with two attempts. I need to reboot my PI before second attempt, couse "already flashing" window pops out if I not.

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/2672463/octoprint.log)
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/2672464/plugin_firmwareupdater_console.log)

benlye, 2018-12-12:
The avrdude path test is not always working, see #65.  If it's not set to `/usr/bin/avrdude` please do so - the leading slash is important.

Zicmu69, 2018-12-12:
TRUE!
I had usr/bin/avrdude in my settings. No way it was that easy... Flashing lika a charm! No more box opening to play with my firm!

Thank you so so much @benlye

benlye, 2018-12-13:
v1.1.0 released which improves the path tests.

closed 2018-12-13 by benlye