OctoPrint/OctoPrint-FirmwareUpdater #67: Flashing Anet Board fails...

Stady234, 2018-12-11:
When i setup to flash:
avrdude
atmega1284p
//usr/bin/avrdude
arduino

i get ::
////////////////////////////////

Flashing failed
Avrdude says: 'not in sync: resp=0xf1
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x4e
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x70
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x7c
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x46
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x93
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x36
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x7a
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x69

avrdude done. Thank you.'
////////////////////////////////

if i change the type to usbasap::
////////////////////////////////

Flashing failed
Avrdude returned code 1
///////////////////////////////

if i change to wiring::
///////////////////////////////

Flashing failed
Timeout communicating with programmer
////////////////////////////////

When i connect to PC i have no issues pushing the Marlin 2.0 to board...

benlye, 2018-12-11:
Try going to advanced settings and setting the baud rate to 57600.

benlye, 2018-12-11:
Also, the correct programmer for the Anet board is 'arduino'.  No need to try the others.  The only variable is the baud rate, which will be either 57600 or 115200.

Stady234, 2018-12-12:
entering 115200 worked on both the Anet board and my Tronxy board !!  Thanks again...

closed 2018-12-12 by Stady234