OctoPrint/OctoPrint-FirmwareUpdater #70: avrdude file path does not exist

Weioo01, 2019-01-08:
Following the instructions from the official github, when I use the stated file path /usr/bin/avrdude and do a test it does not verify. I saw an earlier closed issue but I did not see a solution. I have tried many variants of the file path name. //usr, usr/, etc. No dice and there's very little info out there.

benlye, 2019-01-08:
Have you actually installed avrdude? It is normally in /usr/bin.

If you run `which avrdude` on the command line the response will tell you the precise path to use in the updater config (if avrdude is installed).

Weioo01, 2019-01-08:
> Have you actually installed avrdude? It is normally in /usr/bin.
> 
> If you run `which avrdude` on the command line the response will tell you the precise path to use in the updater config (if avrdude is installed).

I believe you are correct. I did some searching for a couple hours last night and I can't find much documentation on this. On the GITHUB it only explains to install the plugin for Octoprint, which I did. I do not see anything telling me how to install avrdude on a Raspberry Pi / Octoprint. There are some commands shown, but no information on how to get avrdude on the pi.

Sorry I am rather new to everything but I like to think I'm tech savy. A little backstory: I got into this suddenly because after a successful print my Marlin 1.1.9 decided to create an issue and unable to home to Z when auto-homing due to a likely firmware issue. I do not have a laptop but I have Octoprint on the Pi. Printer is far from my desktop so I was looking to upgrade the firmware via Octoprint.

Do I need to connect a KB and monitor to the Pi to run some commands to get avrdude installed? 

Please advise, and thank you!

benlye, 2019-01-08:
"Install avrdude" is listed in the setup instructions:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#setup

Anyway, no need for a keyboard you can do it via SSH with two commands:

`sudo apt-get update`
`sudo apt-get install avrdude`

Weioo01, 2019-01-08:
I saw those instructions, but coming from somebody who doesn't know programming too well, and having to look up how to use SSH, it's vague. It would be very helpful to leave a note on the GITHUB page noting how to connect to talk to the RaspPi in order to submit the noted commands for AVRDUDE installation. I may be putting the horse in front of the carriage as I don't know much about RasbPis other than I needed one to run Octoprint, lol. I plan to follow this guide: https://www.raspberrypi.org/documentation/remote-access/ssh/

Thanks for the help!

closed 2019-01-08 by benlye

Weioo01, 2019-01-08:
I saw those instructions but coming from somebody who is not big into
programming, and having to look up how to use SSH, it's very vague. It
would be very helpful to leave a note on the GITHUB page noting how to
connect to talk to the RaspPi in order to submit the noted commands. I may
be putting the horse in front of the carriage too, I don't know much about
RasbPis other than I needed one to run Octoprint, lol. I plan to follow
this guide: https://www.raspberrypi.org/documentation/remote-access/ssh/

Regardless, thank you!!