OctoPrint/OctoPrint-FirmwareUpdater #72: [Feature Request] Support for Re-Arm

boelle, 2019-01-12:
would be nice if we could flash firmware to the Re-Arm remotely

closed 2019-03-19 by boelle

benlye, 2019-03-19:
@boelle, sorry I never replied to this.  The main reason I didn't get to it is because I don't have the hardware.  If you can submit a PR to add support for that board I'd be happy to include it.

boelle, 2019-03-19:
i closed it as there is no need for it, 

with the re-arm and marlin on it you can expose the onboard sd to the host as a drive.
then it's just a matter of using samba to share that drive over the network and i can upload new firmware.

the only downside is that i have to go to the machine and press the reset switch and then reboot octoprint

not a big deal if i only update once in a while, its more painfull with a new machine where you change settings many times during a day