OctoPrint/OctoPrint-FirmwareUpdater #74: [Feature Request] Firmware flash profiles

Blizzke, 2019-02-22:
Would it be possible to add a way to quickly switch the flashing configuration? I have a a Prusa MK3 with an MMU2 connected to it. Both are also connected to octoprint and both have a firmware. It would be handy if I could flash them with a single profile change instead of having to reconfigure the plugin

matfiz, 2019-07-07:
@Blizzke can you share the profile you're using to upgrade MMU2 through this plugin?

Blizzke, 2019-07-08:
I wrote a script for that. The script allows me to flash the MK2/3 and the MMU2. 

Since the MMU2 can only be flashed for about 5 seconds after it boots IIRC, the script looks for the device, asks you to reset it, locates it again and then starts the flash. You can find it [here](https://gist.github.com/Blizzke/9e13c4a7cbb91ec1976a0c50162fa323)

matfiz, 2019-07-08:
Awesome, @Blizzke! Thank's for sharing. IMHO it is a perfect feature to be incorporated into the FirmwareUpdater.

Tirpitz93, 2021-01-31:
+1 for this feature

benlye, 2021-01-31:
No promises, but I'll look into what this will take to implement this request.  I'm worried about blowing up the number of settings that need to be managed so I need to figure out how to deal with that elegantly.

WarrenSchultz, 2021-02-03:
Thanks! Even two profiles (Printer + MMU) would be great.

trv-in-3d, 2021-02-21:
Another vote for this new feature please.  Ideally the "Profile" would also include the serial port setting.   I love being able to update the MMU and the printer without having to bring in a laptop and switch cables.   Thank you for the new MMU support!

benlye, 2021-02-22:
I'm working on this now.  I'll use this post to track my progress:

- [x] Create a settings structure which supports more than one printer
- [x] Create a process to migrate settings from the old structure to the new one
- [ ] Modify the settings UI to allow printer profiles to be enabled
   - [ ] Make using profiles optional so people with one printer don't have to deal with extra complexity
   - [ ] Use the migrated settings as the default profile
- [ ] Create a way to add/remove/select a printer profile
- [ ] Remember the last-used profile and load it by default
- [ ] Make sure everything works on Python 2 and Python 3

benlye, 2021-02-28:
Progress so far...

Have settings being migrated and the UI updated so that profiles can be selected, added, deleted, copied, and edited.  Need to do some more work on the settings - current plan is to break the settings out into general plugin options (accessed by the wrench icon) and profile settings (accessed by the cog icon).

Am thinking about having an option to disable multiple profiles but am not sure how to implement the settings editor in that scenario.  May scrap that idea, not sure yet.

Feedback or input on the settings / UI is welcome.  I'm just going with my gut at this point.

[image]

benlye, 2021-03-01:
That's a nice example. I'm part way through completely redo ingthe settings UI and it will be tabbed like that. Still working out the details...

benlye, 2021-03-03:
I'm pretty close to a beta version of this.  Here are some screenshots:

Profiles enabled:
[image]

The wrench icon in the top right now only accesses plugin-wide options:
[image]

All flashing options are accessed via the cog icon next to the profile name.  The selected profile is always the one that gets edited.  Settings are now on tabs:
[image]
[image]
[image]
[image]

If profiles are disabled, the wrench icon accesses all settings (like it does in current/previous versions) and all settings are in tabs:
[image]

The serial port is remembered for each profile and when the profile is selected the last-used serial port for that profile will automatically get selected as well, as long as it is present.

Any profile can be edited or renamed, any profile _except the first one_ can be deleted.  Profiles can be copied, in which case all settings are duplicated.

The last few things I need to do are:
* Do a little bit of tidying up in the settings pages
* Test on Python 2

Tirpitz93, 2021-03-04:
looks fantastic

benlye, 2021-03-04:
The first beta with firmware profiles is available on the Development release channel - you have to go into the Software Update plugin and set the release channel for Firmware Updater to Development and you should be able to install v1.10.0

Please read the [release notes](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.10.0).

* This was a major re-write of most of the front-end code and a large re-write of some of the backend.  I've tested as much as I can but there may be some bugs lurking
* I have not tested anything on Python 2 yet

I have tested avrdude, marlinbft, and lpc176x and they all work for me / my printer.

benlye, 2021-03-04:
I found a bug which affected profiles being migrated on Python 2 only.  This has been fixed in [v1.10.1](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/releases/tag/1.10.1) and is available now (in the Devel branch).

benlye, 2021-03-07:
So.... has anyone tried this yet???

trv-in-3d, 2021-03-08:
I tried.  I was on 1.0.9 stable.  Installed 1.10.3 dev.   My previous settings for printer were retained, but I don't have any icons to create a new profile for the mmu.
  
[image]

Did I miss a step somewhere?
I am running octoprint 1.5.3.

benlye, 2021-03-08:
> Did I miss a step somewhere?

Yes, you have to enable profiles. Click the wrench icon, go to the plugin tab, check the enable profiles box.

benlye, 2021-03-17:
Version 1.10.9 released with support for multiple printer / flash profiles.

closed 2021-03-17 by benlye