OctoPrint/OctoPrint-FirmwareUpdater #77: Flashing failed Avrdude returned code 1

codegrue, 2019-03-15:
Can anyone advise how to troubleshoot this? I confirmed avrdude is the latest on the Pi and I can't find a meaning to "code 1" anywhere in any documentation.

benlye, 2019-03-19:
Please attach the logs.

codegrue, 2019-03-19:
When can I find the logs? I was searching already to no avail.

benlye, 2019-03-19:
In OctoPrint go to Settings -> Logging.  Attempt an update and then grab octoprint.log and plugin_firmwareupdater_console.log.

Stoerenfreed, 2019-04-07:
I've got the same problem.. here is my log-file.
what causes this Problem? :/

[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3051447/octoprint.log)

benlye, 2019-04-07:
That log doesn't have the details. Can you attach the firmware updater log, plugin_firmwareupdater_console.log, as well?

Stoerenfreed, 2019-04-07:
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3051590/plugin_firmwareupdater_console.log)

benlye, 2019-04-07:
It looks like you have the wrong combination of board and programmer. Try setting the board to 1284p and the programmer to Arduino. Also, clear the configuration file setting as that's what's causing the Error 1.

Stoerenfreed, 2019-04-07:
Changing the settings to 1284p and arduino didn't solve the problem.

I'm not sure what configuration file you mean or where to find it.  : /

benlye, 2019-04-08:
Under the plugin's advanced settings there is a field for the avrdude configuration file, clear it out. 

Leave the other settings as 1284p and Arduino as the log showed that using the Arduino programmer avrdude was able to talk to your board, but the board reported that it was a 1284p, not a 2560.

Post the firmware updater log again if you still have problems.

Stoerenfreed, 2019-04-08:
That solved it.
Thank you a lot!! : )

closed 2019-04-09 by benlye

sandovalcarlo, 2019-05-27:
I'm having the same issue. I changed my settings to  ATmega1284p and Arduino for the AVR programmer type, and being that it's a no sync issue I thought it may be the bound rate, and I tried them all and it didn't work.

 Here my logs with the last attempt using 115200.

2019-05-27 01:23:43,181 /usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -b 115200 -U flash:w:/tmp/tmpeG5VJR:i
2019-05-27 01:23:43,881 > 
2019-05-27 01:23:43,882 > avrdude: Version 6.3-20171130
2019-05-27 01:23:43,882 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2019-05-27 01:23:43,883 >          Copyright (c) 2007-2014 Joerg Wunsch
2019-05-27 01:23:43,884 > 
2019-05-27 01:23:43,884 >          System wide configuration file is "/etc/avrdude.conf"
2019-05-27 01:23:43,885 >          User configuration file is "/home/pi/.avrduderc"
2019-05-27 01:23:43,885 >          User configuration file does not exist or is not a regular file, skipping
2019-05-27 01:23:43,886 > 
2019-05-27 01:23:43,886 >          Using Port                    : /dev/ttyUSB0
2019-05-27 01:23:43,887 >          Using Programmer              : arduino
2019-05-27 01:23:43,887 >          Overriding Baud Rate          : 115200
2019-05-27 01:23:43,888 > 
2019-05-27 01:23:46,419 > avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x65
2019-05-27 01:23:46,421 > avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
2019-05-27 01:23:46,422 > avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
2019-05-27 01:23:46,423 > avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
2019-05-27 01:23:46,424 > avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
2019-05-27 01:23:46,425 > avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x53
2019-05-27 01:23:46,427 > avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x44
2019-05-27 01:23:46,428 > avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
2019-05-27 01:23:46,429 > avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x63
2019-05-27 01:23:46,431 > avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x61
2019-05-27 01:23:46,432 > 
2019-05-27 01:23:46,433 > avrdude done.  Thank you.
2019-05-27 01:23:46,434 > 

could it be that the boot-loader is corrupted?

benlye, 2019-05-27:
You probably want to keep the baud rate at the default. Can you set it back and post the log?

What board / printer do you have?

tuboil, 2020-04-13:
I'm having the same issue.
> avrdude: invalid record at line 281 of "/tmp/tmph_XEZ7"
file: plugin_firmwareupdater_console.log

2020-04-13 15:34:18,219 
2020-04-13 15:34:18,220 /usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpeu721d:i
2020-04-13 15:34:19,324 > 
2020-04-13 15:34:19,326 > avrdude: Version 6.3-20171130
2020-04-13 15:34:19,327 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2020-04-13 15:34:19,329 >          Copyright (c) 2007-2014 Joerg Wunsch
2020-04-13 15:34:19,330 > 
2020-04-13 15:34:19,331 >          System wide configuration file is "/etc/avrdude.conf"
2020-04-13 15:34:19,332 >          User configuration file is "/home/pi/.avrduderc"
2020-04-13 15:34:19,333 >          User configuration file does not exist or is not a regular file, skipping
2020-04-13 15:34:19,335 > 
2020-04-13 15:34:19,336 >          Using Port                    : /dev/ttyUSB0
2020-04-13 15:34:19,337 >          Using Programmer              : wiring
2020-04-13 15:34:19,338 >          AVR Part                      : ATmega2560
2020-04-13 15:34:19,339 >          Chip Erase delay              : 9000 us
2020-04-13 15:34:19,341 >          PAGEL                         : PD7
2020-04-13 15:34:19,342 >          BS2                           : PA0
2020-04-13 15:34:19,343 >          RESET disposition             : dedicated
2020-04-13 15:34:19,344 >          RETRY pulse                   : SCK
2020-04-13 15:34:19,345 >          serial program mode           : yes
2020-04-13 15:34:19,346 >          parallel program mode         : yes
2020-04-13 15:34:19,347 >          Timeout                       : 200
2020-04-13 15:34:19,349 >          StabDelay                     : 100
2020-04-13 15:34:19,350 >          CmdexeDelay                   : 25
2020-04-13 15:34:19,351 >          SyncLoops                     : 32
2020-04-13 15:34:19,352 >          ByteDelay                     : 0
2020-04-13 15:34:19,353 >          PollIndex                     : 3
2020-04-13 15:34:19,354 >          PollValue                     : 0x53
2020-04-13 15:34:19,356 >          Memory Detail                 :
2020-04-13 15:34:19,357 > 
2020-04-13 15:34:19,358 >                                   Block Poll               Page                       Polled
2020-04-13 15:34:19,360 >            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
2020-04-13 15:34:19,361 >            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
2020-04-13 15:34:19,362 >            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,363 >            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
2020-04-13 15:34:19,364 >            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,366 >            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,367 >            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,368 >            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
2020-04-13 15:34:19,370 >            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
2020-04-13 15:34:19,371 >            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
2020-04-13 15:34:19,372 > 
2020-04-13 15:34:19,373 >          Programmer Type : Wiring
2020-04-13 15:34:19,375 >          Description     : Wiring
2020-04-13 15:34:19,376 >          Programmer Model: AVRISP
2020-04-13 15:34:19,377 >          Hardware Version: 15
2020-04-13 15:34:19,378 >          Firmware Version Master : 2.10
2020-04-13 15:34:19,379 >          Vtarget         : 0.0 V
2020-04-13 15:34:19,380 >          SCK period      : 0.1 us
2020-04-13 15:34:19,382 > 
2020-04-13 15:34:19,383 > avrdude: AVR device initialized and ready to accept instructions
2020-04-13 15:34:19,384 > avrdude: Device signature = 0x1e9801 (probably m2560)
2020-04-13 15:34:19,385 > avrdude: reading input file "/tmp/tmpeu721d"
2020-04-13 15:34:19,386 > avrdude: invalid record at line 281 of "/tmp/tmpeu721d"
2020-04-13 15:34:19,387 > avrdude: read from file '/tmp/tmpeu721d' failed
2020-04-13 15:34:19,389 > 
2020-04-13 15:34:19,390 > avrdude done.  Thank you.
2020-04-13 15:34:19,391 > 
2020-04-13 15:34:19,392 >