OctoPrint/OctoPrint-FirmwareUpdater #83: [Help] Trying to configure settings for Ender 3/Melzi board

NovaViper, 2019-04-15:
I'm trying to setup a configuration that works with the Ender 3 before trying to flash it. Normally within the Audrunio IDE, it's a board configuration of Sanguino (1284p) and the AVR Programmer is AVRISP MKII. How can I setup the same configuration with the plugin?

benlye, 2019-04-15:
Choose Atmega1284p as the MCU and stk500v2 as the programmer.

[image]

In a future update I'll rename 'stk500v2' to 'AVRISP MKII' to make it obvious that they are the same thing.

NovaViper, 2019-04-16:
Thanks for the info! Hey, do you know how I can compile the Marlin firmware to upload?

NovaViper, 2019-04-18:
It doesn't work, it says timeout with programmer <_>

benlye, 2019-04-18:
Please attach the logs. Must likely you are using the incorrect settings for your board.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/README.md#troubleshooting

NovaViper, 2019-04-18:
Sorry about that! Here ya go: 
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3096189/plugin_firmwareupdater_console.log)

benlye, 2019-04-18:
A quick Google says that an Ender 3 needs a bootloader installed before it can be flashed. Have you done that?

Sorry, but this is a printer board I have no experience with. You're probably better off trying to find a Facebook group or other community who can help you to get started.

There also seem to be multiple YouTube videos available. Once you can flash the printer from the Arduino IDE you should be able to use the plugin without problems.

NovaViper, 2019-04-18:
I did flash a boot loader onto the printer, which is even stranger as to why it doesn't work with the plugin

benlye, 2019-04-19:
Are you able to flash it from the Arduino IDE? Can you post the settings you use in the IDE?

NovaViper, 2019-04-19:
I am able to upload through the IDE, these are the settings

Board: Sanguino
Processor: ATmega1284 or ATmega1284p (16hz)
Programmer: AVRISP mkII

NovaViper, 2019-04-19:
This is also the command used by the IDE: -CC:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega1284p -carduino -PCOM5 -b115200 -D -Uflash:w:C:\Users\user\AppData\Local\Temp\arduino_build_474603/Marlin.ino.hex:i

benlye, 2019-04-19:
Yeah, use 'arduino' as the programmer on the plugin, as that's what the IDE is using.

benlye, 2019-04-19:
> This is also the command used by the IDE: -CC:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino14/etc/avrdude.conf -v -patmega1284p **-carduino** -PCOM5 -b115200 -D -Uflash:w:C:\Users\user\AppData\Local\Temp\arduino_build_474603/Marlin.ino.hex:i

To be a little more helpful (sorry, I've been on my mobile the last few days, so more terse than normal)...

The Programmer that you have selected in the Arduino IDE is only used when you choose 'Upload via Programmer'.  When you choose 'Upload' it uses the default programming method for the selected MCU.  You can see in your IDE command line that it is using 'arduino' as that is the '-c' option: `-carduino` .

You don't really have an 'AVRISP MKII' programmer, so there is no use selecting that in the plugin, and it won't work.

In the plugin you should choose board as 'atmega1284p' and programmer as 'arduino' and it *should* work.  If it still doesn't work, upload the firmware updater's log file again.

NovaViper, 2019-04-19:
That fixed it! Thanks 😄

closed 2019-04-19 by NovaViper