OctoPrint/OctoPrint-FirmwareUpdater #85: supported chips

wBrhy2, 2019-04-28:
Hello, for someone who can follow steps to flash devices (whive I`ve been doing for years) but not savvy enough to know the difference between AVR or SAM units, or 1280 and 1286 models.... does my board have to use one of the exact devices listed? I have an AT90USB1286 (printrboard), will it work with on of the AVR options? (since it is an AVR board AFAIK)

Thanks!

benlye, 2019-04-28:
According to the Wiki, it's none of the above and I'd need to add a new option to the menu for you.

https://reprap.org/wiki/Printrboard

However, it seems like there are different ways to flash the board depending on the bootloader that is installed, so I wouldn't be confident without testing it.

Do you know what bootloader you have installed?

wBrhy2, 2019-04-28:
I'll have to confirm the bootloader information when I'm back home. I have successfully flashed using flip in the past, but I moved my printer away from my desktop and my laptop has a chipset unsupported by the universal drivers (god I hate my Dell laptop / tablet).

The fact that flip worked.. Does it mean I had dfu bootloader, or would flip also work on the other bootloaders? Apparently some cards have a lufa cdc....

wBrhy2, 2019-04-28:
Do you have a donation link for btc or eth? Love the plugin.

wBrhy2, 2019-04-28:
Confirmed DFU. It's also the factory default from printrbot (when they still existed, don't know what boards come with now). 

If anyone finds this while searching for information, 3rd party boards could also come with a CDC bootloader.

benlye, 2019-04-29:
I can work with that.  Looks like the Teensy 2.0++ uses the same MCU and bootloader, and is cheap.  I'll get one ordered and when it arrives I'll add support for it.

wBrhy2, 2019-04-29:
That's amazing. So you have a btc donation address?

benlye, 2019-04-29:
I do now ;-)
Bitcoin: `1GjUmcjnAxCr9jFPUtVrr6gPQz8FhYddZz`

Thanks!

wBrhy2, 2019-04-29:
Awesome, sent you a beer ;)

benlye, 2019-04-30:
Thanks, @wBrhy2, I appreciate that.  My Teensy 2.0++ is on its way from China; ETA is at least a couple of weeks.  I'll update the thread here when I've made some progress.

wBrhy2, 2019-05-01:
Even if it doesn't work out, I really appreciate that you are taking the time to dig in like that.

benlye, 2019-05-20:
Brief update: I've received my Teensy++ 2.0 from China and flashed it with the DFU bootloader.  Over the next few days I'll figure out how to flash it from the plugin.

wBrhy2, 2019-05-22:
*so exciting *

benlye, 2019-05-22:
v1.4.0 released with support for Printrboard (or any board with a DFU bootloader, probably).

Board-specific instructions are here:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/README.md#at90usb-boards

Make sure you remove/place the BOOT jumper to get the board into bootloader mode prior to flashing.

wBrhy2, 2019-05-22:
🤯 Amazing! I'll flash later today and let you know! 😁

wBrhy2, 2019-05-22:
does the DFU flasher get installed automatically as a dependency? I am just wondering if I have to do it manually because I did an update instead of a fresh install.

benlye, 2019-05-22:
You have to install dfs-programmer manually. 

You don't need OctoPrint to be connected to flash the firmware, just go to the plugin and kick off a flash. For all the other methods you need to pick a serial port in the flash plugin, but for DFU it's not required.

wBrhy2, 2019-05-22:
just a typo, people won't find dfs-programmer, the package is dfu-programmer ;)

benlye, 2019-05-22:
Make sure you follow the instructions I linked above to install it. The packaged version won't work.

wBrhy2, 2019-05-22:
oh wow, I had somehow missed that! Thanks!

wBrhy2, 2019-05-22:
Flasher works! :D thanks!

benlye, 2019-05-23:
Glad to hear it's working.

closed 2019-05-23 by benlye