OctoPrint/OctoPrint-FirmwareUpdater #91: Disabled Post-Flash delay with empty post-flash delay value results in plug-in reporting failure

qtemp, 2019-06-09:
Flash reports failure when the "Post Flash Delay" setting is empty.  The python script should check to see if the setting is enabled before checking the number of seconds.

Steps:
1. Enable the post-flash delay 
2. Clear the seconds from 0 to empty
3. Disable the post-flash delay
4. Save
5. Flash

**Log Files**
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater/__init__.py", line 227, in _flash_worker
    if float(postflash_delay) > 0 and self._settings.get(["enable_postflash_delay"]):
ValueError: could not convert string to float:

benlye, 2019-06-10:
Ugh.  Thanks for the detailed report.  I'll have a fix out shortly.

benlye, 2019-06-19:
Fixed in v1.5.0.

closed 2019-06-19 by benlye