OctoPrint/OctoPrint-FirmwareUpdater #94: Flashing failed avrdude returned code 1

mister2212, 2019-06-27:
**Hardware type**
Anycubic I3 Mega 8Bit board Marlin 1.1.9
Raspberry 3b+

**Describe the problem**
earlier it was possible to flash a new firmware via octoprint.  with the same settings that worked earlier I now get an error and the process stops.
Flashing failed avrdude returned code 1

**Log Files**
Attach the OctoPrint log and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**

benlye, 2019-06-27:
Can't help without the logs...

benlye, 2019-06-27:
Would also be useful to know what changed on your printer.  I've just checked and I can definitely still flash my boards using avrdude.

mister2212, 2019-06-27:
sorry forgotten. here are the logs

> 2019-06-27 18:29:44,472 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 17.67ms
2019-06-27 18:29:44,497 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2019-06-27 18:29:44,498 - octoprint.server - INFO - OctoPrint 1.3.10
2019-06-27 18:29:44,500 - octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Automatic Shutdown (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_automaticshutdown
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/cura
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Filament Manager (0.5.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager
|  Firmware Updater (1.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmwareupdater
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
| !Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2019-06-27 18:29:44,513 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 917016576
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.15.1
|  python:
|    pip: 19.0.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2019-06-27 18:29:44,514 - octoprint.server - INFO - ------------------------------------------------------------------------------
2019-06-27 18:30:38,977 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-06-27 18:30:39,187 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2019-06-27 18:30:44,639 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.90ms
2019-06-27 18:31:44,619 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.52ms
2019-06-27 18:32:44,651 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.98ms
2019-06-27 18:33:44,527 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 17.50ms
2019-06-27 18:34:44,647 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.57ms
2019-06-27 18:35:44,684 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.67ms
2019-06-27 18:36:44,679 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 10.01ms
2019-06-27 18:37:44,673 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.72ms
2019-06-27 18:38:44,737 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.44ms
2019-06-27 18:39:44,615 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 19.49ms
2019-06-27 18:40:44,580 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 20.90ms
2019-06-27 18:41:44,592 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 18.26ms
2019-06-27 18:42:44,765 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 10.10ms
2019-06-27 18:43:44,776 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 10.63ms
2019-06-27 18:44:44,792 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 10.06ms
2019-06-27 18:45:38,980 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2019-06-27 18:45:39,155 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {}
2019-06-27 18:45:44,780 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.76ms
2019-06-27 18:46:44,654 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 19.52ms
2019-06-27 18:47:44,803 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.69ms
2019-06-27 18:48:44,828 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 9.80ms
2019-06-27 18:49:44,706 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.188.55) 18.18ms
[... cut ...]
2019-06-21 17:55:55,946 > avrdude: Version 6.3-20171130
2019-06-21 17:55:55,946 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2019-06-21 17:55:55,947 >          Copyright (c) 2007-2014 Joerg Wunsch
2019-06-21 17:55:55,947 > 
2019-06-21 17:55:55,947 >          System wide configuration file is "/etc/avrdude.conf"
2019-06-21 17:55:55,947 > avrdude: error at /etc/avrdude.conf:4158: syntax error
2019-06-21 17:55:55,948 > avrdude: error reading system wide configuration file "/etc/avrdude.conf"
2019-06-21 17:55:55,948 > 
`

benlye, 2019-06-27:
At least some of your attempts are using the incorrect programmer.  Keep it set to 'wiring'.

The message says that your arvrdude configuration file has a syntax error in it on line 4158.  I guess if you fix that it will work again.

From your log:
```
2019-06-21 17:07:40,284 > System wide configuration file is "/etc/avrdude.conf"
2019-06-21 17:07:40,285 > avrdude: error at /etc/avrdude.conf:4158: syntax error
2019-06-21 17:07:40,285 > avrdude: error reading system wide configuration file "/etc/avrdude.conf"
```

You might try removing avrdude from your system, deleting /etc/avrdude.conf, then reinstalling avrdude.

mister2212, 2019-06-27:
sorry but can you give me the command for putty to delete avrdude?

closed 2019-06-27 by mister2212