OctoPrint/OctoPrint-FirmwareUpdater #95: [Question] Reboot SKR V1.3 after flash

NovaViper, 2019-07-02:
Hi! I'm wondering how can I reboot my SKR V1.3 after the firmware flashes successfully

benlye, 2019-07-02:
Do you mean you want to do another reboot after the one which flashes the new firmware to the board?  Any particular reason?

You could add `M997` as a post-flash gcode, but that would only get executed when OctoPrint reconnects.

NovaViper, 2019-07-02:
Yeah, normally I would have to turn the power off and back on to get it to load up the new firmware. M997 would do the same thing without having to actually turn off the power?

benlye, 2019-07-02:
The lpc1768 flash routine triggers the reboot by sending M997 to the printer after the file has been copied to the SD card, making the upgrade happen automatically.  Is that not working for you?  What firmware are you currently running?

benlye, 2019-07-02:
Did you see this note in the docs?
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#minimum-marlin-firmware-version

NovaViper, 2019-07-02:
The reset only runs before the flash starts, as per the setting (Reset before flash), it doesn't do anything else after flashing. and I have Marlin 2.0

benlye, 2019-07-02:
That's not what it's supposed to be doing. Can you attach the logs?

NovaViper, 2019-07-02:
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3352375/plugin_firmwareupdater_console.log)

benlye, 2019-07-03:
Could you attach octoprint.log as well?

NovaViper, 2019-07-03:
Here ya go: 
[octoprint.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3356113/octoprint.log)

benlye, 2019-07-03:
The relevant parts of your log are here:
https://pastebin.com/raw/dVFK6Kkj

You've had a success, where the firmware was flashed and the printer successfully reset, and a couple of failures.

You have some others plugins installed which are making a lot of noise, including ipOnConnect.  At some point someone else mentioned that they had an issue with ipOnConnect but never gave me more information (#49).

Could you try disabling ipOnConnect to see if that helps?  Try disabling portlister and navbartemp as well if disabling ipOnConnect doesn't fix it.

NovaViper, 2019-07-04:
I've tried to run it before without that particular plug installed.. but like I said, it never seems to actually 'install' the firmware, it just puts it on the SD card

NovaViper, 2019-07-04:
And it never actually times out, it always puts the firmware on the card but never tells the board/printer to power cycle. In fact I don't recall it ever doing that even with the stock Melzi  Ender 3 board. I've always had to actually power cycle it

benlye, 2019-07-04:
> I've tried to run it before without that particular plug installed.. but like I said, it never seems to actually 'install' the firmware, it just puts it on the SD card

The logs you sent tell a different story.  In the log there are exactly three attempts.  The first succeeded (both the copy and the reset), the second failed to copy the file to the SD card, and the third failed to do the pre-flash reset - it never copied the file either.

> And it never actually times out, it always puts the firmware on the card but never tells the board/printer to power cycle. In fact I don't recall it ever doing that even with the stock Melzi Ender 3 board. I've always had to actually power cycle it

That makes no sense - the Melzi board uses a totally different flashing method (Avrdude), which will always reset the board after an upload.

I can flash my SKR board and others can flash theirs.  I've not heard from anyone else who has this problem, so it must be something specific to your setup.  

We can probably figure it out but you need to be willing to try some things and send more logs.

benlye, 2019-07-04:
By the way, I doubt it's causing any problems, but you should disable or remove your pre-flash M117 command.  The pre-flash and post-flash command fields are for *system* commands.  Gcode won't work there.

NovaViper, 2019-07-04:
Still not working <_>

benlye, 2019-07-04:
Logs?

NovaViper, 2019-07-04:
Huh.. it seemed to have finally decided to actually install the firmware this time around. I went and tested to see if it actually flashed by seeing if the bootscreen wouldn't show up, and it didn't when testing it (since I normally have the bootscreen on.

closed 2019-07-04 by NovaViper

NovaViper, 2019-07-04:
*Facepalm* I just realized that I most likely didn't noticed the firmware had fully flashed the very first time around <_> As I habitually power cycle the printer (since I had done so prior to having the board hooked up to Octoprint whenever I was flashing via my PC). Oh well.. at least I know it works :D