OctoPrint/OctoPrint-FirmwareUpdater #96: AttributeError: 'NoneType' object has no attribute 'rfind' whiletrying to upload

Firzum, 2019-07-02:
Hello,

First of all thanks for this addon !

I'm using latest octoprint with a SKR 1.3 and Marlin 2.0 and I have the following error each time I try to upload the firmeware :

2019-07-02 15:45:49,006 - octoprint - ERROR - Exception on /plugin/marlin_flasher/board/listall [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1245, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1264, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlin_flasher/__init__.py", line 192, in board_listall
    result = arduino.board_listall()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pyduinocli/__init__.py", line 135, in board_listall
    return self.__exec(args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pyduinocli/__init__.py", line 88, in __exec
    p = Popen(command, stdout=PIPE, stderr=PIPE, universal_newlines=True)
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
AttributeError: 'NoneType' object has no attribute 'rfind'

I searched on google without behind able to find a solution. Looks like a variable doesn't exists.
Any idea ?

Thanks in advance.

Regards,

Papy

benlye, 2019-07-02:
Wrong flash plugin.  You need to report that error here:
https://github.com/Renaud11232/OctoPrint-Marlin-Flasher

Alternatively, you're welcome to install and use my plugin instead.

Firzum, 2019-07-02:
Oh sorry, you are right.
And I came here because I had an issue with your plugin that I use, I just mixed the logs in octoprint.log /o\ (I have issues with both addons).
Here is the right log:

2019-07-02 19:38:44,362 - octoprint.plugins.firmwareupdater - ERROR - Firmware folder path is not valid: /media/usb/
2019-07-02 19:38:44,442 - tornado.access - WARNING - 400 POST /plugin/firmwareupdater/flash (::ffff:192.168.1.221) 141.44ms

When I do "test" in configuration part it's OK.

ls -lha /media/usb
lrwxrwxrwx 1 root root 4 Jul  2 15:23 /media/usb -> usb0

ls -lha /media/
total 52K
drwxr-xr-x 10 root root 4.0K Jul  2 15:23 .
drwxr-xr-x 22 root root 4.0K Jun  5 16:32 ..
lrwxrwxrwx  1 root root    4 Jul  2 15:23 usb -> usb0
drwxr-xr-x  2 pi   pi    16K Jan  1  1970 usb0
drwxr-xr-x  2 root root 4.0K Jul  2 15:23 usb1
...

I have no idea on what to do...

Firzum, 2019-07-02:
My bad, I got it, I had setup /media/usb/, remplacing it by /media/usb solved the issue. Sorru for the noise. Maybe this mistake should be considered when the edit box content is read (trim the last "/") and the check also done when the test button is pressed, so it will be more consistent.
Thanks again for your addon !

benlye, 2019-07-02:
> My bad, I got it, I had setup /media/usb/, remplacing it by /media/usb solved the issue. Sorry for the noise. Maybe this mistake should be considered when the edit box content is read (trim the last "/") and the check also done when the test button is pressed, so it will be more consistent.
> Thanks again for your addon !

Yep, I can fix that :-)

Firzum, 2019-07-02:
Thanks again !

closed 2019-07-02 by Firzum