OctoPrint/OctoPrint-FirmwareUpdater #97: STM32 Flash

davidcgu, 2019-07-23:
This pluguin is great!!!!

It was working perfectly for me with my previous board SKR V1.3  LPC1768 however now I moved to an SKR PRO V1.1 board which is STM32 based.

I thought same method than beffore should work, only have changed the port nomenglature from /dev/USB0 to /dev/ttyACM0.

If I test the port on the pluguin it tells path available however when trying to flash directly gives me and error with "flash method is not properly configured"

I don't know exactly why I have this, Any clue?

Thansk in advance

davidcgu, 2019-07-23:
I think just realized is not related to the pluguin, anyway will update just in case it helps someone else

benlye, 2019-07-23:
Highly unlikely that will work with the same method as the other SKR boards with lpc1768 MCUs.  I'll have to look into what it will take to flash it.

davidcgu, 2019-07-23:
I'm sure it will work the same way, in fact I was able to make it run once and flashed succesfully, unfortunately my board is faulty and I can not work with it now anymore

trouch, 2019-07-30:
See https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/98

benlye, 2019-07-30:
@trouch the SKR Pro v1.1 seems like an unusual STM32 board, if the manual can be trusted.

The board's manual says that flashing it is the same as the lpc1768-based SKR v1.1 and v1.3 - copy the firmware to the SD card and reboot.

Being an STM32-based board I would have expected to need to use stm32flash, as you've taken care of in #98, but I'm not sure that's needed for this board.  SKR must have built a custom bootloader to handle flashing.

See page 6 of the manual.
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/manual/SKR%20PRO%20V1.1%20user%20manual%20v1.1.pdf

Not that it matters - having stm32flash as a flash method is cool :-)

trouch, 2019-07-30:
Indeed. I'm not familiar with the SKR Pro but there are definitely different way to flash an STM32 depending on the MCU model and hardware/firmware design:
- DFU
- USB mass storage (lpc1768 like)
- ST serial bootloader (stm32flash)

Having said that, it seems SKR came up with a fourth one using the SD card. Maybe you could use M28/M29 gcode commands to load hex onto the embedded SD card through UART, then reset the board with a UART disconnect/reconnect.

benlye, 2019-07-30:
> Maybe you could use M28/M29 gcode commands to load hex onto the embedded SD card through UART, then reset the board with a UART disconnect/reconnect.

That's a neat idea.  I had an email from someone this morning asking if it was possible to flash via a serial connection because their SKR board had a busted USB port.  I hadn't thought of that - I'll have to play around with it, it might even be easier than using the USB interface to mount the SD card.

benlye, 2019-08-15:
@davidcgu, the problem with your initial attempt is that you had the path to the firmware flash folder set as the serial device.  If you followed the instructions it should have been something like `/media/usb`.

closed 2019-08-15 by benlye

reloxx13, 2019-08-15:
there is no /media/usb (there is nothing mounted).

> If you followed the instructions
ur "hints" in the readme?

benlye, 2019-08-15:
> there is no /media/usb (there is nothing mounted).
> 
> > If you followed the instructions
> 
> ur "hints" in the readme?

More than hints...
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/#lpc1768-boards

reloxx13, 2019-08-15:
i mesnt the stm32 notes not the lpc.   

i got the skr mini e3 with a stm

benlye, 2019-08-15:
Apparantly, and note that I don't have one to test and am only going by what @davidcgu said, the SKR Pro uses the LPC1768 method.

The [doc for the SKR Mini E3](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/tree/master/firmware) says that it also uses the same copy-and-reset procedure, so you should probably also follow the LPC1768 instructions.  

If you still have a problem feel free to open a new issue.

reloxx13, 2019-08-15:
thank you very much! i did not get/understand that. ill give it a try.

benlye, 2019-08-15:
I've added a note for the SKR STM32 boards.  Hopefully this helps clear up any confusion.

[image]