OctoPrint/OctoPrint-FirmwareUpdater #99: LPC1768 use model

marclefevre, 2019-07-30:
I have a Bigtreetech SKR 1.3 (using LPC1768) that seems to be supported. If I configure the settings for that and try an update, the plugin throws an error that it's not configured. I don't know where else to configure it.
Does the update consist of copying the firmware file to the SD card that the controller presents to octoprint?

benlye, 2019-07-30:
Yes, that's pretty much how it works.

Can you attach a screenshot of your plugin settings?

marclefevre, 2019-07-30:
[image]
[image]
[image]

benlye, 2019-07-31:
You have specified a serial port where you should have put the path to the SD card mount point.
[image]

If you followed the [LPC1768 instructions ](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/README.md#lpc1768-boards)and installed USBmount, the path should be `/media/usb`, **not** `/dev/ttyACM0`.

marclefevre, 2019-07-31:
OK.  Thanks for the pointer.  I mistakenly assumed that the process was simple enough that there were no instructions.  I followed them and have usbmount installed.  I have gotten the reset failed error and am working on getting the latest Marlin.
One suggestion I might make is to detect an attempt to set the folder to a device file and, rather than confirm it's valid, point out that it needs to be a directory.
Thanks for all of the help.

closed 2019-07-31 by marclefevre

benlye, 2019-08-16:
> One suggestion I might make is to detect an attempt to set the folder to a device file and, rather than confirm it's valid, point out that it needs to be a directory.

Fixed in v1.6.1.