OctoPrint/OctoPrint-FirmwareUpdater #101: Skr 1.3 Flashing does not occur

jimmyeao, 2019-08-04:
**Hardware Setup**
BigtreeTech SKR1.3

**Describe the problem**
Flashing does not occur, even though Plugin reports successful
**Log Files**
Attach the OctoPrint log
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3465084/plugin_firmwareupda
[plugin_firmwareupdater_console.log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3465086/plugin_firmwareupdater_console.log)
[octoprint (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3465087/octoprint.1.log)
[serial (1).log](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3465088/serial.1.log)

ter_console.log)
g and the Firmware Updater log Log files, which can be accessed under **Settings** -> **Logging** in OctoPrint, or in `~/.octoprint/logs/` on the host.  Both `octoprint.log` and `~/.octoprint/logs/plugin_firmwareupdater_console.log` should be included.

**Additional context**
Examing the SD card after a flashe shows only a firmware.cur file with an old date.

benlye, 2019-08-04:
I'm wondering if you have the board's SD card mounted correctly.  I can see that the firmware.bin file is getting copied to /media/usb0 - are you sure that this is the right path, and that the board's SD Card is mounted there?  What does `ls -al /media/usb0` show?

jimmyeao, 2019-08-04:
```

pi@octopi:~ $ ls -al /media/usb0
total 212
drwxr-xr-x  3 pi   pi    16384 Jan  1  1970 .
drwxr-xr-x 10 root root   4096 Jul  7 19:23 ..
-rw-rw-rw-  1 pi   pi   164304 Aug  4 18:16 FIRMWARE.CUR
drwxr-xr-x  2 pi   pi    16384 Jun 24 19:56 System Volume Information
pi@octopi:~ $ ls -al /media/usb
lrwxrwxrwx 1 root root 4 Jul  7 19:23 /media/usb -> usb0
pi@octopi:~ $
```

benlye, 2019-08-04:
Strange.  The update process is very simple - copy the file to the SD card and reset the board.  It looks like the reset is working (and you would know if it wasn't), so the only part that can be broken is the copy.

I've just tested on my SKR board and v1.6.0 of the plugin is working fine and, in any case, there were no changes to the lpc1768 code in v1.6.0.

The thing I find interesting in your log is this line:
`Line 12592: 2019-08-04 17:20:12,819 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb0' available for writing after 0.0 seconds`

On my system it takes around 10 seconds for usbmount ot re-mount the SD card after a pre-flash reset.  That your Pi thinks it's available immediately makes me wonder if it really is there, or if the plugin is writing to an empty mount point.

Does it behave any differently if you disable the pre-flash reset?  If you try that, can you confirm that the board's SD card is really mounted before you try the flash?

benlye, 2019-08-04:
Also, the check that the SD Card is mounted is pretty basic - it looks to see if the mount point is writable.  If, for some reason, your OctoPrint system is able to write to /media/usb0 before anything is mounted there, then that would be why it fails to wait for the SD card to really be mounted.  Root can do this.

Are you running OctoPrint as root, pi, or some other user?

Can you do `ls -al /media/` and post the output?

jimmyeao, 2019-08-04:
pi@octopi:/ $ ls -al /media/
total 64
drwxr-xr-x 10 root root  4096 Jul  7 19:23 .
drwxr-xr-x 21 root root  4096 Nov 13  2018 ..
lrwxrwxrwx  1 root root     4 Jul  7 19:23 usb -> usb0
drwxr-xr-x  3 pi   pi   16384 Jan  1  1970 usb0
drwxr-xr-x  3 pi   pi   16384 Jan  1  1970 usb1
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb2
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb3
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb4
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb5
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb6
drwxr-xr-x  2 root root  4096 Jul  7 19:23 usb7

jimmyeao, 2019-08-04:
Interstingly, it seems there is an issue accessing the card:

```
pi@octopi:/ $ touch /media/usb/test.txt
touch: cannot touch '/media/usb/test.txt': Input/output error
pi@octopi:/ $ touch /media/usb0/test.txt
touch: cannot touch '/media/usb0/test.txt': Input/output error
```

jimmyeao, 2019-08-04:
Flashing has put the .bin file on the card now,

```
pi@octopi:~ $ ls -al /media/usb0
total 180
drwxr-xr-x  2 pi   pi     4096 Jul  8 13:31 .
drwxr-xr-x 10 root root   4096 Jul  7 19:23 ..
-rw-r--r--  1 pi   pi   173628 Aug  4 20:01 firmware.bin

```
I got an error if i unticked reset board before upload, so I left that ticked. However, it hasnt actually flashed the firmware, just put it on the card ready (from what I can see)

Flashing a second time seemed to apply it:
```
2019-08-04 20:07:03,093 - octoprint.plugins.firmwareupdater - INFO - Executing pre-flash commandline 'M117 Flashing Firmware'
2019-08-04 20:07:03,120 - octoprint.plugins.firmwareupdater - INFO - Pre-flash command 'M117 Flashing Firmware' returned: 32512
2019-08-04 20:07:03,121 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2019-08-04 20:07:03,124 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb0'
2019-08-04 20:07:03,163 - octoprint.plugins.firmwareupdater - INFO - Pre-flash reset: attempting to reset the board
2019-08-04 20:07:03,163 - octoprint.plugins.firmwareupdater - INFO - Resetting LPC1768 at '/dev/ttyACM0'
2019-08-04 20:07:03,212 - octoprint.plugins.firmwareupdater - INFO - Waiting for LPC1768 at '/dev/ttyACM0' to reset
2019-08-04 20:07:04,757 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' is resetting
2019-08-04 20:07:07,972 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.93) 25.06ms
2019-08-04 20:07:08,008 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' reset in 4.79 seconds
2019-08-04 20:07:08,010 - octoprint.plugins.firmwareupdater - INFO - Release the firmware lock on the SD Card by sending 'M22' to '/dev/ttyACM0'
2019-08-04 20:07:08,056 - octoprint.plugins.firmwareupdater - INFO - Waiting for SD card to be available at '/media/usb0'
2019-08-04 20:07:08,058 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb0' available for writing after 0.0 seconds
2019-08-04 20:07:08,065 - octoprint.plugins.firmwareupdater - INFO - Copying firmware to update folder '/tmp/tmpZpNe6c' -> '/media/usb0/firmware.bin'
2019-08-04 20:07:08,069 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb0'
2019-08-04 20:07:08,105 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2019-08-04 20:07:08,106 - octoprint.plugins.firmwareupdater - INFO - Resetting LPC1768 at '/dev/ttyACM0'
2019-08-04 20:07:08,155 - octoprint.plugins.firmwareupdater - INFO - Waiting for LPC1768 at '/dev/ttyACM0' to reset
2019-08-04 20:07:12,567 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' is resetting
2019-08-04 20:07:15,807 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' reset in 7.65 seconds
2019-08-04 20:07:15,812 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.
2019-08-04 20:07:15,823 - octoprint.plugins.firmwareupdater - INFO - No postflash gcode or postflash is disabled, setting run_postflash_gcode to false

```
and I now have a FIMRWARE.CUR on the card
although now the card is mounted as sdb1 and is again corrupt.
Suspect a faulty card?
```
[ 9829.988855] FAT-fs (sdb1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
pi@octopi:/dev $ sudo fsck.vfat -nVv /dev/sdb1
fsck.fat 4.1 (2017-01-24)
Checking we can access the last sector of the filesystem
0x25: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.
Boot sector contents:
System ID "        "
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
     16384 bytes per cluster
         1 reserved sector
First FAT starts at byte 512 (sector 1)
         2 FATs, 16 bit entries
     15360 bytes per FAT (= 30 sectors)
Root directory starts at byte 31232 (sector 61)
       512 root directory entries
Data area starts at byte 47616 (sector 93)
      7676 data clusters (125763584 bytes)
0 sectors/track, 0 heads
         0 hidden sectors
    245728 sectors total
Starting check/repair pass.
Checking for unused clusters.
Starting verification pass.
Checking for unused clusters.
Leaving filesystem unchanged.
/dev/sdb1: 5 files, 14/7676 clusters

```

jimmyeao, 2019-08-04:
Update.
It hasn't flashed, but what is odd is the time stamps:

```
pi@octopi:~ $ ls -al /media/usb0
total 180
drwxr-xr-x  2 pi   pi     4096 Aug  4 20:03 .
drwxr-xr-x 10 root root   4096 Jul  7 19:23 ..
-rw-r--r--  1 pi   pi   173628 Aug  4 20:07 firmware.bin
-rw-r--r--  1 pi   pi        0 Aug  4 20:03 test.txt
pi@octopi:~ $ ls -al /media/usb0
total 212
drwxr-xr-x  3 pi   pi    16384 Jan  1  1970 .
drwxr-xr-x 10 root root   4096 Jul  7 19:23 ..
-rw-rw-rw-  1 pi   pi   164304 Aug  4 18:16 FIRMWARE.CUR
drwxr-xr-x  2 pi   pi    16384 Jun 24 19:56 System Volume Information
```
Firmware.cur pre-dates firmware.bin :/

benlye, 2019-08-04:
> Flashing has put the .bin file on the card now,
> 
> ```
> pi@octopi:~ $ ls -al /media/usb0
> total 180
> drwxr-xr-x  2 pi   pi     4096 Jul  8 13:31 .
> drwxr-xr-x 10 root root   4096 Jul  7 19:23 ..
> -rw-r--r--  1 pi   pi   173628 Aug  4 20:01 firmware.bin
> ```
> 

That can't be the contents of the SD card.  There should at least be a FIRMWARE.CUR file there as well.

Maybe the permissions on the folders in /media are messed up.  Can you unplug the board, reset the Pi, and then do `ls -al /media/`?

It should look similar to this:
```
pi@pi3:/media $ ls -al /media/
total 44
drwxr-xr-x  11 root root 4096 Apr 11 19:03 .
drwxr-xr-x  22 root root 4096 Nov 21  2017 ..
lrwxrwxrwx   1 root root    4 Apr 11 19:03 usb -> usb0
drwxr-xr-x   2 root root 4096 Apr 11 20:56 usb0
drwxr-xr-x   2 root root 4096 Aug  4 18:40 usb1
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb2
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb3
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb4
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb5
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb6
drwxr-xr-x   2 root root 4096 Apr 11 19:03 usb7
```

benlye, 2019-08-04:
> Update.
> It hasn't flashed, but what is odd is the time stamps:
>
> ```
> pi@octopi:~ $ ls -al /media/usb0
> total 180
> drwxr-xr-x  2 pi   pi     4096 Aug  4 20:03 .
> drwxr-xr-x 10 root root   4096 Jul  7 [...]

 I think you're seeing the content of the folder before the SD card has mounted (where you see firmware.bin), then later the content of the folder after the SD has mounted (where you see FIRMWARE.CUR.

I'm thinking that the permissions are messed up on your mountpoints and they probably need to be reset.

jimmyeao, 2019-08-04:
Could be, but that's an almost 2 hour difference :)
I'll refer to your earlier post and reset everything shortly and try again!

benlye, 2019-08-04:
With the board unplugeed, I would try this:
```
sudo chown root:root /media/usb0
sudo chown root:root /media/usb1
```

Then plug the board in and give it a few seconds to mount.  Then you should see this:
```
pi@pi3:/media $ ls -al
total 104
drwxr-xr-x  11 root root  4096 Apr 11 19:03 .
drwxr-xr-x  22 root root  4096 Nov 21  2017 ..
lrwxrwxrwx   1 root root     4 Apr 11 19:03 usb -> usb0
drwxr-xr-x   2 pi   pi   65536 Jan  1  1970 usb0
drwxr-xr-x   2 root root  4096 Aug  4 18:40 usb1
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb2
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb3
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb4
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb5
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb6
drwxr-xr-x   2 root root  4096 Apr 11 19:03 usb7
```

benlye, 2019-08-04:
The crucial thing is that the pi user mustn't have write access to the mount point until *after* the SD card is mounted (which usbmount takes care of), but it seems like on your system that is not the case.  Resetting the permissions should take care of that.

jimmyeao, 2019-08-04:
Cool, will give that a shot in the morning and report back, thanks for the help :)

jimmyeao, 2019-08-05:
> With the board unplugeed, I would try this:
>
> ```
> sudo chown root:root /media/usb0
> sudo chown root:root /media/usb1
> ```
>
> Then plug the board in and give it a few seconds to mount. Then you should see [...]

Thats fixed it, many thanks!

closed 2019-08-05 by jimmyeao

jimmyeao, 2019-08-05:
for completeness:
```
2019-08-05 17:01:26,166 - tornado.access - WARNING - 403 GET /api/job (::ffff:192.168.0.146) 10.98ms
2019-08-05 17:01:30,276 - octoprint.plugins.firmwareupdater - INFO - Firmware update folder '/media/usb0' available for writing after 12.0 seconds
2019-08-05 17:01:30,277 - octoprint.plugins.firmwareupdater - INFO - Copying firmware to update folder '/tmp/tmpnZ7Mli' -> '/media/usb0/firmware.bin'
2019-08-05 17:01:30,912 - octoprint.plugins.firmwareupdater - INFO - Unmounting SD card: 'sudo umount /media/usb0'
2019-08-05 17:01:30,986 - octoprint.plugins.firmwareupdater - INFO - Firmware update reset: attempting to reset the board
2019-08-05 17:01:30,987 - octoprint.plugins.firmwareupdater - INFO - Resetting LPC1768 at '/dev/ttyACM0'
2019-08-05 17:01:31,083 - octoprint.plugins.firmwareupdater - INFO - Waiting for LPC1768 at '/dev/ttyACM0' to reset
2019-08-05 17:01:31,960 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' is resetting
2019-08-05 17:01:34,393 - tornado.access - WARNING - 409 GET /api/printer (::ffff:192.168.0.93) 23.91ms
2019-08-05 17:01:36,831 - octoprint.plugins.firmwareupdater - INFO - LPC1768 at '/dev/ttyACM0' reset in 5.74 seconds
2019-08-05 17:01:36,834 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.
2019-08-05 17:01:36,847 - octoprint.plugins.firmwareupdater - INFO - No postflash gcode or postflash is disabled, setting run_postflash_gcode to false
2019-08-05 17:01:36,853 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyACM0, baudrate=115200, profile={'heatedChamber': False, 'name': 'Ender 3 Pro SKR1,3', 'color': 'default', 'axes': {'y': {'speed': 3000, 'inverted': False}, 'x': {'speed': 3000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 300, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 220.0, 'width': 220.0, 'custom_box': False, 'height': 250.0}, 'model': 'Creality Ender 3 Pro', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
```