OctoPrint/OctoPrint-FirmwareUpdater #102: [Feature request] Allow to combine with github actions

TheZoker, 2019-08-13:
I don't know if that is a good idea, but my plan is the following:
I have a fork of marlin, where I have my own configuration as a commit and rebase it regularly (to keep it up2date with the marlin repo).
All these changes are automatically tested via the new github actions (so essentially: `platformio run build`).
Now my idea is, if the build was successfully, the new firmware should be uploaded to octroprint and flashed automatically to the printer. That would save me a lot of time (basically that`s continuous deployment).
So a step within the actions could look like:
```
...
      - id: build
        name: Build
        run: platformio run
      - id: auto_flash
        name: Automatically flash firmware
        uses: OctoPrint/action-octoprint@1.0.0
        env:
          OCTOPRINT_IP: ${{ secrets.OCTOPRINT_IP }}
          OCTOPRINT_API_KEY: ${{ secrets.OCTOPRINT_API_KEY }}
        if: job.steps.build.status == success
...
```

So with that, every time I rebase my config onto the latest marlin firmware and it builds successfully, the firmware will be updated automatically. 

And in combination with this plugin, this would be fantastic. 
So the request would be:
- Allow to upload firmware to octoprint from action
- Allow to flash this firmware remotely (secured with the API key)

benlye, 2019-08-15:
This is an interesting idea, but pretty niche - I can't see many people using it yet it would take a fair bit of time to implement.  I have to be honest, I'm unlikely to do it.

If you wanted to figure it out and submit a pull request I'd be happy to consider it.

NovaViper, 2019-08-23:
@TheZoker How did you manage to setup your branch? I've been looking into this myself aswell

TheZoker, 2019-08-23:
@NovaViper You can find my current workflow here: https://github.com/TheZoker/Marlin/blob/ender-3/.github/workflows/main.yml

But it does not work:
https://github.com/TheZoker/Marlin/commit/50fbf2b049dbeb2cf40bb337e97945ef3eac6bd4/checks#step:5:14

I'm not sure how to fix it, to be honest

NovaViper, 2019-08-24:
How did you add the workflow option in your repo? @TheZoker

TheZoker, 2019-08-24:
You can signup for the beta here: https://github.com/features/actions

TheZoker, 2019-08-29:
@NovaViper I fixed the workflow and now it works.
My config is here: https://github.com/TheZoker/Marlin/blob/ender-3/.github/workflows/main.yml
And an example run is here: https://github.com/TheZoker/Marlin/commit/628bbd99c14bbecf803d1081e5519a46fe58b654/checks

NovaViper, 2019-09-02:
@TheZoker Nice! Though I still haven't gotten into the beta DX

Is there some other way of getting it to work?

TheZoker, 2019-09-03:
Not if you want to use github actions. But you can still use travis for that:
https://github.com/TheZoker/Marlin/blob/ender-3/.travis.yml

NovaViper, 2019-09-04:
@TheZoker Thanks for the info! I just got the email for joining Github actions. I'm wondering if you made a made a completely seperate branch or just used one of the existing ones

NovaViper, 2019-09-30:
@TheZoker Hey what is the SLACK_WEBHOOKS string for? That's what keeps throwing errors for me

closed 2020-12-05 by github-actions[bot]