OctoPrint/OctoPrint-FirmwareUpdater #103: [FR] Updating SKR mini E3

reloxx13, 2019-08-15:
**Hardware Setup**
SKR mini E3 in an Ender 3

**Describe the problem**
Currently, its not possible to mount the SD Card due Marlin.

benlye, 2019-08-15:
So are you saying that the lpc1768 method doesn't work for the SKR Pro boards after all?

The documentation for the SKR Mini E3 and SKR Pro certainly sound like the lpc1768 copy-and-reset method should work.

Is there something I can change to make it work?  I don't have a board to test with so am forced to rely on what people are telling me...

reloxx13, 2019-08-15:
sry, i mismatched, skr pro 1.1 is also under stm, ill look again if i see some differences.

right now there is not usb mount upcoming, no blockdevice with
sudo blkid -o list

benlye, 2019-08-15:
Have you edited your Configuration_adv.h file and set `#define SDCARD_CONNECTION ONBOARD`?

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration_adv.h#L1044

reloxx13, 2019-08-15:
yeeep :S

```
  #if HAS_SDCARD_CONNECTION
    /**
     * Set this option to one of the following (or the board's defaults apply):
     *
     *           LCD - Use the SD drive in the external LCD controller.
     *       ONBOARD - Use the SD drive on the control board. (No SD_DETECT_PIN. M21 to init.)
     *  CUSTOM_CABLE - Use a custom cable to access the SD (as defined in a pins file).
     *
     * :[ 'LCD', 'ONBOARD', 'CUSTOM_CABLE' ]
     */
    #define SDCARD_CONNECTION ONBOARD
  #endif
```

and also added
```

//
// SD Support
//
#ifndef SDCARD_CONNECTION
  #define SDCARD_CONNECTION ONBOARD
#endif
```

in the pins file

benlye, 2019-08-15:
Do you have any other computer you can plug it into?  Try a different cable?  If it doesn't show up as a storage device then you have some other issue going on.  Either that or this board simply doesn't expose the SD card over USB...

reloxx13, 2019-08-15:
ill try it.

i just dont wanted to spam the other issue :D

if someone searches for mini3 this will come up with eitehr a it will work or it wont work.

transfering the firmware by gcode would still be an option.

benlye, 2019-08-15:
It might be worth asking on the BigTreeTech SKR Mini repository if SD card access over USB is expected to work:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues

I don't think that the update by gcode will work - I don't think it would handle binary data.

reloxx13, 2019-08-15:
on windows i only get a com connection with 3 cables

benlye, 2019-08-15:
> on windows i only get a com connection with 3 cables

Right, so you need to find out if that's normal for your board, or if you can use some combination of Marlin firmware options to make it appear.

I'm going to remove the SKR Mini E3 comments from the readme until you find out.  For sure it will not work with either the STM32 flash method or the LPC1768 method.

reloxx13, 2019-08-15:
i did write an issue and a fb to btt asking this.

p3p, 2019-08-15:
Although SKR added the ability to flash from the SD card to their bootloader they haven't added a USB MSC device to their Arduino framework (or the ststm32 platform, at least not by default like my lpc176x framework)

Hopefully I can stabilise the Marlin Binary Protocol in the next week or so, supporting it (if you can from within octoprint, and think its worth the effort) would allow you to transfer binary data over the serial connection onto the sd card.

benlye, 2019-08-15:
> Hopefully I can stabilise the Marlin Binary Protocol in the next week or so, supporting it (if you can from within octoprint, and think its worth the effort) would allow you to transfer binary data over the serial connection onto the sd card.

Sounds good - I'll certainly look at it.

benlye, 2019-08-15:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3-/issues/14

p3p, 2019-08-15:
Just for reference (https://github.com/MarlinFirmware/Marlin/pull/14817), I am moving the client responses to binary but with that change it will hopefully stabilise the transport layer and sd card transfer protocol.

omahena, 2019-08-27:
I have an SKR Pro 1.1... Aaaand there is (currently) no way to expose the SD card to Marlin or make it available through an USB connection. Their documentation is extremely poorly written and for example links Marlin build instructions for SKR 1.1 in the SKR Pro 1.1 manual (note the missing Pro). Similar... But very different.

One of the problems is that it looks like the SPI bus the SD card uses is in conflict with various other board features and pins. Unfortunately even after removing all the conflicts I could find... It is still unable to send any data to the SD card.

So there must be some other problems present as well. For now I couldn't find them.

If anyone here has a SKR Pro 1.1 with a working SD card, please let me know. I am looking for inspiration on what could be wrong with the SD card so I'd love to compare notes.

@benlye I don't think you can do anything to make this work. But nice work on the plugin. I love the idea of not having to walk to the printer and fiddling with the SD cards to update the firmware. My floor is starting to show signs of wear from all my walking to the printer...

reloxx13, 2019-09-07:
BTT added a PR to marlin to get SD Mounting working 👍    
(not merged now)   
https://github.com/MarlinFirmware/Marlin/pull/15180

falo2k, 2019-11-17:
Any luck since the BTT PR was merged?

CRCinAU, 2019-12-20:
Yeah - I'm curious on how this is going :)

reloxx13, 2019-12-20:
works fine now if you build marlin firmware with usb support.

falo2k, 2019-12-22:
Very cool, will try it later.  Need to change anything from the defaults for stm32flash?

reloxx13, 2019-12-28:
@Rmkiv4 

**OctoPrint on a Pi with Rasbian and FirmwareUpdater Plugin Settings for SKR mini E3 v1**

- Be sure to flash the USB Version of the firmware. You can choose this if you build the FW yourself with Visual Studio Code.

[image]

[image]

The firmeware file will now be copied to the mounted sd card and a M997 command will be send to restart the printer. The Bigtreetech bootloader on the skr board will now find the new firmware on the sd card and flash it.

I wrote a script sh to backup the whole SD Card before/after flashing. 
It will also delete old files.

Prep:
mkdir -p /home/pi/.octoprint/scripts/fw_backup/
vim /home/pi/.octoprint/scripts/fw_backup/fw_backup.sh
Hit i and paste the script below
Hit ESC
Enter `:wq` and hit enter to write and quit vim
sudo chmod +x /home/pi/.octoprint/scripts/fw_backup/fw_backup.sh to make the file executable.

Update FirmwareUpdate Pre- or Post -flash Settings to run the script before/after flashing flashing

Preflash
`sudo mount /dev/sda1 /media/usb1 && /home/pi/.octoprint/scripts/fw_backup/./fw_backup.sh`

Postflash (dont forget to tick the checkbox to enable Postflash script
`/home/pi/.octoprint/scripts/fw_backup/./fw_backup.sh`

```
#!/bin/bash

BACKUPDIR="/home/pi/.octoprint/scripts/fw_backup/backups"
DATE="$(date +%Y%m%d_%H%M%S)"

#echo "$BACKUPDIR/$DATE"

mkdir -p "$BACKUPDIR/$DATE"

rsync -rlpgoDv --exclude "System Volume Information" /media/usb1/ "$BACKUPDIR/$DATE/"

find "$BACKUPDIR" -mindepth 1 -maxdepth 1 -type d -iname "*" -mtime +30 -exec rm -rf {} \;
```

closed 2019-12-28 by reloxx13

cropduster, 2020-02-04:
> yeeep :S
>
> ```
>   #if HAS_SDCARD_CONNECTION
>     /**
>      * Set this option to one of the following (or the board's defaults apply):
>      *
>      *           LCD - Use the SD drive in the external LCD [...]

If I check in platform.io, the section      #define SDCARD_CONNECTION ONBOARD in configuration_adv.h is greyed ourt, so it indicates it is inactive. Why is that? I have #define SDSUPPORT active in coinfiguration.h.

CRCinAU, 2020-02-04:
I never got this working - however, this is related:
https://github.com/MarlinFirmware/Marlin/issues/16509

cropduster, 2020-02-04:
I was never able to mount the Sd card. I have connected a raspberry pi via usb cable, running repetier server, and I would like to be able to copy over a firmware.bin to the sd card via the USB cable, so I can issue M997 to restart the printer and update the firmware without exchanging sd cards.

cropduster, 2020-02-04:
Do I need the STM32F103RC_bigtree_512K_USB board defined in platform.io for this to work or is it supposed to work also with the STM32F103RC_bigtree_512K ?

reloxx13, 2020-02-04:
> 
> 
> Do I need the STM32F103RC_bigtree_512K_USB board defined in platform.io for this to work or is it supposed to work also with the STM32F103RC_bigtree_512K ?

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/103

Deses, 2020-02-07:
> > Do I need the STM32F103RC_bigtree_512K_USB board defined in platform.io for this to work or is it supposed to work also with the STM32F103RC_bigtree_512K ?
> 
> #103
> [image]

Newer versions of BTT Marlin have this:
[image]

And we are supposed to use `STM32F103RC_bigtree_512K`.

reloxx13, 2020-02-07:
> > > Do I need the STM32F103RC_bigtree_512K_USB board defined in platform.io for this to work or is it supposed to work also with the STM32F103RC_bigtree_512K ?
> > 
> > 
> > #103
> > [image]
> 
> Newer versions of BTT Marlin have this:
> [image]

this is an older marlin version.
go for latest marlin and start with the example config for your board from marlin config repo.

Deses, 2020-02-07:
Uh oh, sorry for the misunderstanding, then!

On another note, I'm now trying to follow what you did here: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/103#issuecomment-569449404

But I don't have sda1... Octoprint finds the Printer's SD card no problem, though.
```
sudo mount /dev/sda1 /media/usb1
mount: /media/usb1: special device /dev/sda1 does not exist.
```

frenchbeast, 2020-04-05:
I was able to get the card mounted 👍🏼
Hope the following helps.

!! Use **FAT** to format your MicroSD card
> ExFAT won't work

Also added to /etc/fstab
```
/dev/sda1       /media/sdcard          auto       defaults,sync,noauto,umask=000  0  1
```
platformio.io
```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_btt_512K_USB
```

[image]

JeremyKennedy, 2020-06-11:
Just wanted to add how I got this to work. Everything mentioned by @frenchbeast was great, I just had to do one more thing - create the mount point.

1. Change board to USB type in `platformio.io`: `STM32F103RC_btt_512K_USB`
2. Rebuild and flash firmware the manual way, so that the next steps can properly mount the SD card as USB
3. Add entry to `/etc/fstab`: 
```
/dev/sda1       /media/sdcard          auto       defaults,sync,noauto,umask=000  0  1
```
4. Create the mount point: `sudo mkdir /media/sdcard`
5. Test this by mounting manually with: `sudo mount /dev/sda1 /media/sdcard`
6. If that worked, reboot the Pi and confirm it's auto-mounted: `ls /media/sdcard`
7. Configure FirmwareUpdater to use the `lpc1768` method, specify and test the path
8. I didn't include the Post-flash delay since the wait seemed unnecessary.

The LPC1768 steps in this plugin's README using `usbmount` also worked, but I reverted those changes in favor of this approach.

I think it would be good to include these steps in the README, and update it so that it no longer states this board is unsupported.

benlye, 2020-06-13:
@JeremyKennedy could you do a PR for changes to the README?  Also, would your steps work for all lpc1768 boards, or is this just for the SKR Mini E3?

sdaitzman, 2020-06-21:
I can confirm this works for me on an SKR Mini E3 DIP, compiled for STM32F103RC_bigtree_512K_USB.

Firmware for reference: [sdaitzman/marlin](https://github.com/sdaitzman/marlin)

Sekisback, 2020-09-05:
I can confirm this works for me on an SKR Mini E3 too. Did it exactly like from JeremyKennedy summarized. this made my day, no more pulling out the sd-card on my BTT SKR mini E3

Deses, 2020-09-05:
My SD card for some reason doesn't automount on boot, even though the fstab entry is fine... I think?

```
proc                  /proc           proc    defaults          0       0
PARTUUID=6c586e13-01  /boot           vfat    defaults          0       2
PARTUUID=6c586e13-02  /               ext4    defaults,noatime  0       1
/dev/sda1             /media/sdcard   vfat     defaults,sync,noauto,umask=000  0  1
```
And testing the path tells me that 'Path to firmware folder /media/sdcard is not writeable'.

I'm going to try to format the sd card, maybe it corrupted itself.

Deses, 2020-09-06:
I can also confirm! I used the usbmount method and it works wonderfully.

This is amazing, finally I can use this plugin. Thank you @JeremyKennedy!!

CRCinAU, 2020-09-06:
Does mounting the SD Card from the Pi still cause stuttering with prints?

The big issue prior wasn't that this couldn't work - but that the USB stack in Marlin wasn't being serviced properly and it was a massive task to try and fix it at the SDK level...

Sekisback, 2020-09-06:
> 
> 
> Does mounting the SD Card from the Pi still cause stuttering with prints?
> 
> The big issue prior wasn't that this couldn't work - but that the USB stack in Marlin wasn't being serviced properly and it was a massive task to try and fix it at the SDK level...

not in my case, i run a pi3b with lasst octoprint version and marlin 2.0.6.1
no stuttering or some strange behavior on my ender3 the last print took 8 hours and the surface is smooth as expected

Deses, 2020-09-06:
Be advised: When running USB composite, things like running a PID autotune from Octoprint will trigger a connection error.

https://github.com/MarlinFirmware/Marlin/issues/15993

CRCinAU, 2020-09-07:
> Be advised: When running USB composite, things like running a PID autotune from Octoprint will trigger a connection error.
> 
> [MarlinFirmware/Marlin#15993](https://github.com/MarlinFirmware/Marlin/issues/15993)
> 
> So you can't PID through console and remote update at the same time...

I thought this was the case....

Deses, 2020-09-07:
You can always do the PID autotune using the TFT screen and then update the values in Marlin if you wish, but they get saved in the EEPROM anyway. If that's the only drawback of enabling USB composite I can live with it, if more problems arise I'll just revert back to good ol' `STM32F103RC_btt_512K`.

TBJ, 2020-09-07:
I can't get automount to work here. I followed the instructions posted by @JeremyKennedy but it wont mount automatically. I release the card with M22 and no matter how long I wait, nothing shows up in the /media/usb.

If I do a manual mount with the same command that umount uses:
`mount -tvfat -osync,noexec,nodev,noatime,nodiratime,gid=pi,uid=pi,dmask=0022,fmask=0111 /dev/sda1 /media/usb0`
Then the SD card mounts no problem and I can move a firmware file manually with an SCP client and trigger an M997 through Octoprint and it successfully flashes. 

However it would be nice to get the automount to work if anyone has any ideas what it may be?

tablatronix, 2020-09-10:
I see this is closed, is this working on SKR E3 v2.0 ?
Ill mess with it tomorrow.

Readme still says

> Not-yet Supported Boards
> SKR Mini E3 - waiting on more information, see this issue

bakaufman, 2020-09-21:
Naive question: does the existing firmware need the be the _USB variety on the E3 mini for future OTA updates through octoprint, or just the nascent version?

sdaitzman, 2020-09-22:
@bakaufman I had to turn on the _USB variant iirc. Here's my firmware for reference: https://github.com/sdaitzman/Marlin

sdaitzman, 2020-09-22:
https://github.com/sdaitzman/Marlin/blob/2.0.x/platformio.ini#L21

calini, 2020-10-23:
@TBJ +1 here; Got an SKR Mini E3 v2.0 with the stock screen, manually mounting the SD card works, but it does not want to automount no matter what. set `SDCARD_CONNECTION` to `ONBOARD`, and I'm running the latest `2.0.x-bugfix` as of today.

o5, 2020-10-29:
I can confirm that @calini 's guide works, thanks! A little suggestion:

ad 1. Change `TM32F103RC_btt_512K_USB` => `STM32F103RC_btt_512K_USB`
ad 3. Remove backticks arround name of the package

calini, 2020-10-30:
@o5 Thanks for the corrections.

Zuckme, 2020-11-25:
So I got the SKR mini v2.0 mini working no problem with this plugin. My host is a FreeBSD box so I will not release the details because I think you all are using a raspi, soooo... I would like anyhow to share my experience.

1. **USE a xxx_USB eniviroment to build your firmare.bin**

As many here have already pointed out, you need to use a _USB enviroment (ie. STM32F103RC_btt_512K**_USB**) to be able to upgrade the the firmware.
In other words by doing so, the SD card on the board can be accessed by the USB interface as a FAT USB disk.
First step after flashing the board with the _USB option, check if the SD card can be written over the USB interface by using a windows box or whatever you like. If you are not able to write and read the SD card on the SKR MINI with a generic PC, your problem is not FirmwareUpdater!
Use only small size SD card! The one delivered with the board 128MB works perfectly. It must be a small size card otherwise the FAT file system can not play well on it. 

2. **Make sure the Octoprint host system (ie. Raspi) can write on the USB disk (SD)**

At this point we have done the job at the SKR MINI and we need to concentrate about the operating system hosting Octoprint.
Forget about SD card and firmware.bin for a moment. The next step now is to make Octoprint able to write the connected USB disk. Yes it is just a FAT USB disk no more no less. Here there are tons material over the web just google it.

Mandatory it to test the location where you mounted the USB disk by using the button "Test":
[image]
see the green "The path is valid" line? If not probably it is not a FirmwareUpdater bug but a wrong settings in the Octoprint host operative system. Use for example SSH and play around to understand why Octoprint can not write where you mounted the USB disk. In my case it was always something wrong at the operative system level and not a FirmwareUpdater bug.

Pay attention that if you are using a Laptop or a PC Box the BIOS can detect the USB disk when it is booting up and do not load the operating system at start up. In this case you need to play with the BIOS Boot priority settings.

3. **DONE, try it out now**

At this point for FirmwareUpdater you are just a user with a lpc1768 board. Forget about the SKR mini compatibility or not.
If step 1 and 2 were successful I dare to say the you have a SKR mini compatible with FirmwareUpdater. Period and congrats.
Well now it is the time to test the firmware update, here my configuration:

[image]

I don't like the automount wankery... I want to control what is mounted and what is not mounted on my box.
Load you firmware.bin and give it a try... It should work.

**Expert corner**
The Bigtreetech guys suggest to copy the firmware.bin on the sd card and then restart the system. If we look at the driver on github, there is a DFU device [here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/mapleDrv-USBdriver/dfu). DFU means Device Firmware Update, see for example [here](http://www.atmel.com/dyn/resources/prod_documents/doc7745.pdf). Soo some guy with a brain bigger than me should be able to update the firmware directly without writing anything on the SD card. Anyway it is not documented nor suggested by Bigtreetech, have fun you dragons...

BTW:
Can anybody confirm me the Post-flash command is before the M997 command? Thanks.

Now it is time for me to see if it is stable or not this plug in.

Have fun

benlye, 2020-11-27:
@Zuckme any chance you could do a PR with documentation for this?

I don't have any of these SKR boards and a lot has been said in this thread. It would be great if someone who has a board can write some concise instructions that can be added to the documentation.

Zuckme, 2020-11-27:
> PR with documentation

@benlye 
Since I have a FreeBSD box, I think my documentation would not help the 99% users who are running your plugin on a Raspi.
I tried with my above post to describe all the generic steps as simple as possible.

I think after all what is wrote here everybody can configure it correctly.

I suggest to link [this](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/103#issuecomment-642830604) as a PR.
It is the best.

holocronology, 2020-12-02:
Are the SKR Mini E3 boards working now?

sdaitzman, 2020-12-02:
@holocronology worked for me w/ the instructions above on an SKR E3 DIP. I just set the settings and set up an usb automount and it worked 😀

Zuckme, 2020-12-02:
Once you get the SD on USB (see above) the SKR Mini E3 is not different than any other LPC1768 board.
Yes it is working here too.

tablatronix, 2020-12-02:
So you have to flash via usb?, I was hoping to get rid of usb altogether..

sdaitzman, 2020-12-02:
> So you have to flash via usb?, I was hoping to get rid of usb altogether..

Hmm, can you say more about what you mean by getting rid of USB altogether? I think you'd need some form of connection to flash it.

tablatronix, 2020-12-02:
Yeah uart