OctoPrint/OctoPrint-FirmwareUpdater #105: Allow flashing of Prusa MMU2

Mirarkitty, 2019-08-17:
**Hardware Setup**
Raspberry pi 3 with Prusa MK3S with MMU2S.

**Describe the problem**
Firmware updater doesn't seem to (easily) support flashing of the secondary firmware that resides on the MMU2 board. 

These is the avrdude command:

avrdude -v -p atmega32u4 -c avr109 -P <virtual serial port> -b 57600 -D -U flash:w:MM-control-01.ino.hex:i

(from https://github.com/prusa3d/MM-control-01/blob/master/README.md)

It would be very nice if the firmware updater could flash both the main unit and the MMU2. Both can be connected to the pi at all times.

Pitta, 2019-08-25:
Wouldnt this require tethering the MMU2 via USB as well?

Mirarkitty, 2019-08-25:
Yes. It comes with the appropriate cable and my rpi3 has ports over...

ftamminga, 2019-09-19:
I wouldn't mind switching cables. 
Also a compatibility test would be nice. Tonight I accidentally flashed the MMU firmware to the MS3S with nasty results.

trv-in-3d, 2019-09-21:
Another vote for this feature.     The prusa MMU has a usb port on it.   I would hope for a solution that would operate as follows:

Setup:
- Leave the printer and the MMU connected to the pi at all times on two different USB ports.
- Your add-in would scan all serial ports and get some kind of id for each board(s) attached.
- allow user to select the found port(s) and configure the setting - and hopefully provide a "name".
Flashing:
- I can select the "name" to upload to, and the bin file to flash with.
- if possible, add-in would make sure expected (board type?, id?) was found on selected serial port before flashing to help prevent flashing wrong device.

Failing that I wouldn't mind switching cables also - but it would be nice if I could create two "profiles" in the add-in for settings - assuming the board types are different.   Then the process would be:
- switch cables
- select correct profile
-select bin file to flash with.

SixFive7, 2020-12-01:
Another vote. Also I wouldn't mind using my rig as a test bench if need be.

MFHFozzy, 2020-12-05:
I would love this. Flashing my MMU is a PITA.

matfiz, 2020-12-05:
This: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/74 gives a lot of hope!

MFHFozzy, 2020-12-05:
This + [#105](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/105) is the dream, i think.

Samaelson, 2021-01-13:
Hey guys, did some rework on the firmware updater by adding MMU2 support.

Can be found here: https://github.com/Samaelson/OctoPrint-FirmwareUpdater/blob/master/README.md

CW1 support and further improvements coming next.

benlye, 2021-01-13:
> Hey guys, did some rework on the firmware updater by adding MMU2 support.
> 
> Can be found here: https://github.com/Samaelson/OctoPrint-FirmwareUpdater/blob/master/README.md
> 
> CW1 support and further improvements coming next.

Are you planning on submitting a pull request for this, or are you going to be publishing and maintaining your own plugin?

Samaelson, 2021-01-14:
I'm open to marge this to original firmware updater also.

CW1 support is added already.

benlye, 2021-01-14:
> I'm open to merge this to original firmware updater also.
> 
> CW1 support is added already and code is rebased to 1.7.6

Would be great if you could rebase on the devel branch and submit a PR (to devel).

Samaelson, 2021-01-14:
OK, will do that in short.

Btw. file patching of  Prusa MMU and CW1 file as well as full CW1 support is added now, also.

Samaelson, 2021-01-14:
PR issued

benlye, 2021-01-14:
Thanks @Samaelson 👍 

PR merged and a beta release tagged.  You can install using the OctoPrint updater by setting the plugin's release channel to 'Development' and installing v1.7.7b3. 

Please try it out.

Samaelson, 2021-01-15:
Tried v1.7.7b3 and works for me

benlye, 2021-01-21:
v1.7.7 released with support for flashing Prusa MMU and CW1.

closed 2021-01-21 by benlye

Tirpitz93, 2021-01-31:
+1 for the option to have multiple profiles to select from, this would allow easy flashing of both the printer and MMU unit.