OctoPrint/OctoPrint-FirmwareUpdater #107: Python 3 compatible

sl1pkn07, 2019-08-23:
**Hardware Setup**
none at tis time, only test

**Describe the problem**
since octoprint (devel branch) is compatible with python3, this plug should be also compatible. the plugin is installed throught plugin manager (inserting url)

**Log Files**
~~~
ago 23 19:33:29 sL1pKn07 sh[17813]: | *Firmware Updater (1.6.1) = /opt/octoprint-git/lib/python3.7/site-packages/octoprint_firmwareupdater

ago 23 19:33:29 sL1pKn07 sh[17813]: Prefix legend: ! = disabled, # = blacklisted, * = incompatible
~~~

~~~
ago 23 19:33:29 sL1pKn07 sh[17813]: 2019-08-23 19:33:29,815 - octoprint.environment - INFO - Detected environment is Python 3.7.4 under Linux (linux). Details:
ago 23 19:33:29 sL1pKn07 sh[17813]: |  hardware:
ago 23 19:33:29 sL1pKn07 sh[17813]: |      cores: 48
ago 23 19:33:29 sL1pKn07 sh[17813]: |      freq: 2200.0
ago 23 19:33:29 sL1pKn07 sh[17813]: |      ram: 67488968704
ago 23 19:33:29 sL1pKn07 sh[17813]: |  os:
ago 23 19:33:29 sL1pKn07 sh[17813]: |      id: linux
ago 23 19:33:29 sL1pKn07 sh[17813]: |      platform: linux
ago 23 19:33:29 sL1pKn07 sh[17813]: |  python:
ago 23 19:33:29 sL1pKn07 sh[17813]: |      pip: 19.2.2
ago 23 19:33:29 sL1pKn07 sh[17813]: |      version: 3.7.4
ago 23 19:33:29 sL1pKn07 sh[17813]: |      virtualenv: /opt/octoprint-git
~~~

**Additional context**
Add any other context about the problem here.

onovy, 2020-03-12:
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/pull/121

benlye, 2020-03-31:
Merged and v1.7.0 released..  Updated version will be available to Py3 Octoprint installations once https://github.com/OctoPrint/plugins.octoprint.org/pull/421 is merged into the plugin repo.

closed 2020-03-31 by benlye