OctoPrint/OctoPrint-FirmwareUpdater #109: Problems with Pi 4 and SKR 1.3

Key0nee, 2019-08-30:
**Hardware Setup**
OctoPrint on Raspberry Pi 4 (Buster) followed [these instructions](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337)
After 3 days of printing, tweaking, etc... this is the only issue I've found...

After following the instructions to get the FirmwareUpdater working with LPC1768 based boards, Im having problems making SKR 1.3 onboard SDCard accessible for OctoPrint to Mount to then use OctoPrint-FirmwareUpdater to make changes.

while the SDCard is healthy and I have no problems with the card this is what dmesg (last few lines ) reports:

[ 3122.935683] sd 0:0:0:0: [sda] Asking for cache data failed
[ 3122.935697] sd 0:0:0:0: [sda] Assuming drive cache: write through
[ 3130.045984]  sda: sda1 sda2
[ 3130.050583] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 3133.623637] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 3200.528869] sd 0:0:0:0: [sda] 124735488 512-byte logical blocks: (63.9 GB/59.5 GiB)
[ 3210.208142] sd 0:0:0:0: [sda] 124735488 512-byte logical blocks: (63.9 GB/59.5 GiB)

I think it has something to do with permissions as this line appears in my OctoPrint.log file

IOError: [Errno 13] Permission denied: u'/media/usb0/.testballoon.txt'

I thought permissions are controlled by this line in my USBMount.conf file

FS_MOUNTOPTIONS="-fstype=vfat,gid=pi,uid=pi,dmask=0022,fmask=0111"

I feel that maybe its permissions at a lower lever than set here but im not sure where to look or how to set.    I am also not sure that marlin bugfix2.0.x is configured correctly to support this.  

any help is greatly appreciated ....

all question, comments, an plain wise crack remarks are welcomed.

thank you 
K

benlye, 2019-08-30:
Most likely the printer's SD Card is not actually mounted.  Can you confirm that the board's SD card is properly mounted?  You should see files in /media/usb/.

Key0nee, 2019-08-30:
Thanks for the reply....
Sorry I am in a bit over my head....

Im not sure what you mean here... If I plug in the SKR directly to my Mac, I do see the card and it appears to be properly mounted.

but when trying to reach from a terminal window it is not. I've tried using this line as both pi and root...

mount /dev/sda1 ~/SDCard/
or
sudo mount /dev/sda1 ~/1SDCard/

again Im in a a bit over my head but wouldn't the line below from my log file confirm that the sd card is in fact not mounted? 

IOError: [Errno 13] Permission denied: u'/media/usb0/.testballoon.txt'

not sure where to go from here.

Thanks again
K

benlye, 2019-08-30:
When you configure the Firmware Updater plugin you must specify a path to the Firmware update folder - what did you put there?

On your Pi, can you run these commands and post the output:
`ls -al /media/`
`ls -al /media/usb/`

It doesn't really matter what it does on your Mac, we're only worried about the Pi.  If you followed all the instructions exactly it should have worked, so we just need to figure out where something is not right...

Key0nee, 2019-08-30:
pi@raspberrypi:~ $ ls -al /media/
total 40
drwxr-xr-x 10 root root 4096 Aug 29 20:32 .
drwxr-xr-x 21 root root 4096 Aug 30 09:31 ..
lrwxrwxrwx  1 root root    4 Aug 29 20:32 usb -> usb0
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb0
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb1
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb2
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb3
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb4
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb5
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb6
drwxr-xr-x  2 root root 4096 Aug 29 20:32 usb7
pi@raspberrypi:~ $ ls -al /media/usb
lrwxrwxrwx 1 root root 4 Aug 29 20:32 /media/usb -> usb0
pi@raspberrypi:~ $

benlye, 2019-08-30:
Right, the board is not mounted - when there's nothing mounted the mount points are read-only, which is why OctoPrint tells you that it could not write to /media/usb0.

Is there anything else in /var/log/messages indicating why usbmount failed?

Can you go through the usbmount installation again and make sure you followed all the steps?

Key0nee, 2019-08-30:
Ok... Ill recheck my USBMount installation...  

/var/log/syslog   list this:

Aug 30 12:50:43 raspberrypi usbmount[3773]: executing command: mount -tvfat -osync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 /dev/sda1 /media/usb0
Aug 30 12:50:43 raspberrypi kernel: [16062.016298] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

benlye, 2019-08-30:
Well we can see that at least it's trying, which is good.  You could always try running the mount command manually:

`sudo mount -tvfat -osync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 /dev/sda1 /media/usb0`

That will only mount the board once, it won't be re-mounted after a disconnect or reboot.  If it errors it might help track down the issue.

Key0nee, 2019-08-30:
redid my USBMount install following all the instructions... still no luck

tried to mount manually as suggested above with:

pi@raspberrypi:~ $ sudo mount -tvfat -osync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 /dev/sda1 /media/usb0

this is what was retuned:
mount: /media/usb0: special device /dev/sda1 does not exist.

I am confident that my USBMount install is correct... however I am not confident that Marlin.bugfix2.0 downloaded 8-19-19 is properly configured to support this plugin.

**from** the instructions for this plugin... could you elaborate on: 

> _The Marlin board configuration must have SDCARD_CONNECTION ONBOARD (previously USB_SD_ONBOARD) enabled so that the on-board SD card is presented to the host via the USB connection. **This seems to be the default configuration for Marlin's LPC1768 boards. It is configured in the board's pins file**._

Im a bit confused as it looks like in marlin bugfix2.0.x this setting seems to be done in the Configuration_adv.h file and by default is set to:

SD_CARD_CONNECTION LCD

Key0nee, 2019-08-30:
I guess I was too quick to state the USBMount was install correctly.... well, I still believe it is installed correctly but it appears that there are problems with Buster....  as stated [here](https://github.com/rbrito/usbmount/issues/25) 

after reading the referenced thread I managed to get the plugin working with Udevil

while I lack the skill to make permanent via a startup script, I managed to get it working by mounting manually with 

udevil mount /dev/ssd1

the sdcard shows up right away in /media/pi/....   and the path is validated by the plugin and able to upload newly compiled firmware files...

Thank You

benlye, 2019-08-31:
Ahh, OK.  I'll add a note to the docs.  Glad you got it working :)

talldonkey, 2019-10-05:
I ran into exactly this - usbmount doesn't seem to work on Raspberry Pi, despite attempting to add PrivateMounts=no in systemd-udevd

`[Service]
MountFlags=shared
PrivateMounts=no`

talldonkey, 2019-10-05:
Got it working.  

What I had to do is follow this exactly: https://github.com/rbrito/usbmount/issues/25#issuecomment-518341969

So, 

1. Install Usbmount via the deb package using wget (see link)
2. I did not have to mess with FS_MOUNTOPTIONS.  I left FS_MOUNTOPTIONS="" as default
3. Then Run 
sudo systemctl daemon-reload
sudo service systemd-udevd --full-restart
I didn't have to mess with systemd-udevd.

Hopefully someone else can verify this from scratch, and then we can update docs

NovaViper, 2020-01-10:
Im having the same issue with the Raspberry  Pi 3B+ running Octopi 0.17

NickClark, 2020-01-23:
Following the existing directions except, instead of 
```
[Service]
MountFlags=shared
```

use 

```
[Service]
PrivateMounts=no
```
Works now for me

benlye, 2020-03-12:
Documentation has been updated.

closed 2020-03-12 by benlye

ybjlus, 2020-05-21:
I also have the same problem running a pi2B. Tried also reinstalling usbmount via the deb package using wget. But no luck.. When I run: ` sudo mount -tvfat -osync,noexec,nodev,noatime,nodiratime,gid=pi,dmask=0022,fmask=0111 /dev/sda1 /media/usb0` 
I get "does not exist" I also tried changing to: ```
[Service]
PrivateMounts=no
```
Any more ideas?

HaernCalistinson, 2020-06-06:
I was able to get this working with the most recent version of OctoPi (buster) and an SKR 1.4 Turbo by setting #define `SDCARD_CONNECTION LCD` instead of `ONBOARD` in Configuration_adv.h

When it was set to `ONBOARD` I could never get it to mount.