OctoPrint/OctoPrint-FirmwareUpdater #110: Failure to mount SD on pi with recent marlin 2.0.x firmwares

Kokorone, 2019-10-09:
**Hardware Setup**
SKR 1.3 in a Creality CR10 but printer independent, tested on Ender 3 Pro

**Describe the problem**
SD card will not mount and instead have Disk I/O errors with recent marlin 2.0.x firmware (10/5/2019 and newer tested, 9/19/2019 version confirmed working)

**Log Files**
Can't even get to the point where octo starts logging anything because sd card doesnt mount, meaning I cant specify folder path line /media/usb0. Provided is a screenshot of the error context 
[image]

**Additional context**
Checked USBMount configurations as stated in FirmwareUpdater setup guide and verified everything is correct.

benlye, 2019-10-10:
If new Marlin builds are breaking it, you probably want to open an issue on the Marlin Github repo.

Have they changed the USB config options again?

benlye, 2019-10-10:
Never mind, I see you did already.
https://github.com/MarlinFirmware/Marlin/issues/15500

Kokorone, 2019-10-11:
It appears to be fixed after updating my usbmount to the dev version from the link here

[https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/109#issuecomment-538698635](url)

closed 2019-10-11 by Kokorone