OctoPrint/OctoPrint-FirmwareUpdater #112: LPC1768 with Buster requires different method

Jademalo, 2019-11-08:
> 3. Configure systemd-udevd so that the mount is accessible
> 
>    `sudo systemctl edit systemd-udevd`
>    
>    Insert these lines then save and close the file:
>
>    `MountFlags=shared`
> 
>    Then run:
>    ```
>    sudo systemctl daemon-reload
>    sudo service systemd-udevd --full-restart
>    ```
> 

The command to edit udevd doesn't work anymore, and the flags have been changed slightly.
According to [here]( https://raspberrypi.stackexchange.com/questions/100312/raspberry-4-usbmount-not-working/100375#100375), it should be changed to the following;

> 
> 3. Configure systemd-udevd so that the mount is accessible
> 
>    `sudo nano /lib/systemd/system/systemd-udevd.service`
>    
>    Ediit the following line then save and close the file:
>    ```
>    PrivateMounts=no
>    ```
> 
>    Then run:
>    ```
>    sudo systemctl daemon-reload
>    sudo service systemd-udevd --full-restart
>    ```

Jademalo, 2019-11-10:
Actually, I can't get this to work either. No matter what I do, Re-ARM's SD card doesn't get mounted. It's definitely enabled in firmware too.

Jademalo, 2019-11-10:
Right, I've figured out why the SD card wasn't getting mounted - It was being mounted by Marlin itsself. Doing an M22 released it and it immediately automounted.

chrisqwertz, 2019-11-14:
For me it was necessary to edit `/lib/systemd/system/systemd-udevd.service` and change `PrivateMounts=yes` to `PrivateMounts=no` there to get the SD card to finally show up on /media/usb

Jademalo, 2019-11-14:
Interesting, it definitely worked adding it to the extra file that you get with `systemctl edit systemd-udevd` for me. Both ultimately do the same thing at the end of the day, just one is editing it directly and another is adding to it afterwards.

ShiJuuRoku, 2019-11-16:
For me the SD card is mounted into /media/usb0 ..

Jademalo, 2019-11-16:
`/media/usb` should be an alias for `/media/usb0`

ShiJuuRoku, 2019-11-16:
Ok.
With this command `ls -R /media` i have :
[image]

hamster65, 2020-02-22:
My mount fails because the dirty bit gets always set when the MKS SGEN Board is reset during the previous firmware update. I am using FAT32 on the sd card. Should I use a different file system?
But even with a "clean" fs, I cannot get it to mount with automount. I have to do this by hand.

benlye, 2020-04-13:
usbmount options have been updated on the doc page to reflect the changes needed for Buster.

closed 2020-04-13 by benlye