OctoPrint/OctoPrint-FirmwareUpdater #113: Pre-flash GCODE command support

lrybak, 2019-11-19:
Would it be possible to implement pre-flash GCODE command support?
Every time I'm flashing my SKR 1.3 I have to manually enter M22 command to release SD card in order to allow mount the filesystem on the octopi OS. Thanks

benlye, 2019-11-20:
That's odd - the LPC1768 flash routine used by the SKR 1.3 already sends an M22 command to the printer before flashing starts.

https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/octoprint_firmwareupdater/methods/lpc1768.py#L193

Can you check your logs to see if this is succeeding?

Heikoho, 2020-03-12:
Hi,
i have the same issue
Is there a workaround or the possibility to pre-send the M22 command?

Thanks

benlye, 2020-03-12:
The reason I didn't add pre-flash G-Code initially is that it assumes/requires the printer to be connected.  I can add it, but will need to think about how it should work if the printer isn't connected when the flash is started.

I've been neglecting this project for a little while, but will get back on this and the other requests soon!

Heikoho, 2020-03-12:
Hi,

thank you for your answer.

 

ok, i wait.

 

Heiko

 
 

Gesendet: Donnerstag, 12. März 2020 um 15:47 Uhr

benlye, 2020-11-26:
So I know this feature request has been open for over a year - sorry!  I have this ready to go but an looking for some feedback.

* For (hopefully) obvious reasons I have implemented the feature so that commands are only sent if the printer is connected when the firmware updated is triggered.  If the printer isn't connected nothing is sent before the flash is started.
* I also added a pre-flash gcode delay because OctoPrint doesn't wait to see if the command was successful before moving on.  This lets you add a brief delay so that whatever command you run can complete.

Does that sound OK?  @Zuckme, does that address the #153 for you?

benlye, 2020-11-26:
Example:
[image]

```
2020-11-26 20:10:28,787 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-11-26 20:10:28,788 - octoprint.plugins.firmwareupdater - INFO - Sending pre-flash gcode commands: M115; M117 FW update starting
2020-11-26 20:10:28,790 - octoprint.plugins.firmwareupdater - INFO - Pre-flash delay: 3s
2020-11-26 20:10:28,813 - octoprint.plugins.z_probe_offset - INFO - Printer_cap[eeprom]:1
2020-11-26 20:10:28,819 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-11-26 20:10:28,825 - octoprint.plugins.z_probe_offset - INFO - Printer_cap[z_probe]:1
2020-11-26 20:10:31,795 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-11-26 20:10:31,872 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-11-26 20:10:31,881 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-11-26 20:10:31,907 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m2560 -c wiring -P /dev/ttyACM0 -D -V -U flash:w:/tmp/tmpg3ngsz8z:i' in /usr/bin
2020-11-26 20:10:33,056 - octoprint.plugins.firmwareupdater - INFO - Writing memory...
2020-11-26 20:10:58,826 - octoprint.plugins.firmwareupdater - INFO - Flashing successful.
2020-11-26 20:10:58,836 - octoprint.plugins.firmwareupdater - INFO - Setting run_postflash_gcode flag to true
```

Zuckme, 2020-11-27:
> Does that sound OK? @Zuckme, does that address the #153 for you?

I think the Pre-flash GCODE is a nice feature you should implement asap. Unfortunately it does not address my point.
I have a SKR mini and it does not need a  M22 command as for the SKR 1.3 to get the job done.

In other words before you unplug a USB drive you should always umount it and with this plugin there is no way I can do that.
Everytime I upgrade the firmware I copy the firmware.bin on the attached USB disk (aka SD card) and then brutally unplug the USB device from the host system by executing a M997. This is not an elegant way at all.

It is a real problem? I do not know.
Will it cause sporadically file system corruption on the SD card? I do not know.

All I can say it is not the proper way to manage a USB disk.

**My proposal**

Instead to use custom boxes and keep in the user interface a separation beetween GCODE and host operating system commands before and after, keep it simple, let the user do what it wants all in one custom field.
Of course we need a way to distinguish between GCODE and commands, what about to put a "#GCODE" Tag before the Marlin code? And why not put a #EXE_FIRMWARE do perform the firmware action the user loaded?

Instead to clutter the user interface with all this input boxes (before and after, GCODE and command and delay), implement just one big expert flash mode box!
The users will now have the freedom to do what they want and less headaches/issues/feature requests for you.

Example of expert flash mode:

```
#GCODE M22
#GCODE M117 M22 sent
mount -t msdosfs /dev/da0s1 /mnt/sd_3dp
#GCODE M117 Mount done
sudo mount /dev/sda1 /media/usb1; /home/pi/.octoprint/scripts/fw_backup/./fw_backup.sh
#GCODE M117 Backup before flash done, starting flash
#EXE_FIRMWARE
#GCODE M117 Firmware sucessfully updated to SD CARD
umount /mnt/sd_3dp
#GCODE M117 Umount done, fire M997 up
#GCODE M997
/home/pi/.octoprint/scripts/fw_backup/./fw_backup.sh
#GCODE M117 Backup after flash done
#GCODE M115
```

You get the idea I hope, you need to implement something like this

```
for code_line in exp_input_box:
     if code_line.startwith('#GCODE'):
           sent_gcode(code_line.replace('#GCODE ', '')
           # go to next line
           continue
     if  '#EXE_FIRMWARE' in code_line:
           call_firmware_action(firmware_file)
           # go to next line
           continue
     # command line
     subprocess.Popen(code_line) 
```

Regarding delays, if the user need some delay they can use `G4` at Marlin level or i.e. `sleep` at command level.
Yes it's tricky, but hey it's expert mode, they need to know what they are doing.

benlye, 2020-11-27:
I think I'll implement this as-is.

If you need to run a system command before M997, why aren't you using pre-flash command? It's already there and should do what you want, no?

benlye, 2020-11-27:
I've released v1.7.1 with pre-flash gcode commands.

@Zuckme fee free to take the conversation about your request back to #153 if using the existing pre-flash system command option doesn't do what you need.

closed 2020-11-27 by benlye