OctoPrint/OctoPrint-FirmwareUpdater #114: SKR PRO FirmwareUpdater Fails

DroneMang, 2019-12-10:
**Hardware Setup**
SKR PRO, RPi 3B+, FT-6, standard up to date Octoprint, 12-7-2019 Marlin 2.0 bugfix

**Describe the problem**
I am trying to upload firmware to the SKR PRO board.  
I have followed: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater

With everything I have tried, nothing ever shows up as mounted with: 
ls -al /media/
ls -al /media/usb/

and the path test in the octoprint plugin never works.

I have turned on these in Marlin 2.0:
#define SDCARD_CONNECTION ONBOARD
#define SDSUPPORT

I have tried updating usbmount as below and described in the links:
mkdir ~/tmpu && cd ~/tmpu
wget https://github.com/nicokaiser/usbmount/releases/download/0.0.24/usbmount_0.0.24_all.deb
sudo dpkg -i usbmount_0.0.24_all.deb

I have tried omitting:
[Service] MountFlags=shared

or changing it to:
[Service] MountFlags=shared PrivateMounts=no

I heard that Lino Barreca was working on this as well but I'm not sure what he has done that is different than this plugin
[Marlin-bugfix-2.0.x-7.zip](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/files/3942460/Marlin-bugfix-2.0.x-7.zip)
: https://github.com/LinoBarreca/Marlin

Links:
https://github.com/rbrito/usbmount/issues/25
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/109#ref-issue-511982242
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/109
https://ignorantguru.github.io/udevil/udevil--help.html
https://github.com/IgnorantGuru/udevil
https://ignorantguru.github.io/udevil/

Hopefully this can work somehow!
-DroneMang

BastR, 2019-12-22:
Hi, maybe you can try again, Lino Barreca has pushed all this work in this PR https://github.com/MarlinFirmware/Marlin/pull/16260

It compiles correctly with SPI drivers and I am able to access to the onboard SD card files from the LCD as well as from octoprint.

LinoBarreca, 2020-01-07:
@boelle @DroneMang This isn't a bug related to here and should be closed.
Let me explain why.
To update the firmware on the SKR-Pro we have several methods.
The easiest (and most user-friendly) one is to copy the firmware.bin on the SKR-Pro onboard card and then trigger a reboot.

This can actually be achieved through my pr (which enabled the on board sdcard) and the M28 B1 command (which must be enabled in configuration). You should open a bug to octoprint firmware updater (or octoprint) to support M28 B1.

LinoBarreca, 2020-01-07:
@DroneMang
To update the firmware on the SKR-Pro we have several methods.
The easiest (and most user-friendly) one is to copy the firmware.bin on the SKR-Pro onboard card and then trigger a reboot through M997

This can actually be achieved through my pr (which enabled the on board sdcard) and the M28 B1 command (which must be enabled in configuration). The bug should be "Enable support M28 B1 command to write rom binaries for flash update"

benlye, 2020-03-12:
The instructions have been updated with new options for the usbmount service.
https://github.com/OctoPrint/OctoPrint-FirmwareUpdater#usbmount-installation

vover000, 2020-04-01:
Help me solve the problem.
The fact is that when assembling a 3dprinter, I accidentally shorted the lower part of the SKR PRO V1.1 Board with a wire, resulting in the burning of the stm32f407zgt6 Microcontroller that is installed on it, I found and bought a new Stm32f407zgt6, replaced it.
Now the problem is to flash the correct bootloader file.bin.
Simply put, you need the correct bootloader file.bin for this Microcontroller.
The skr pro motherboard itself was not affected by a short circuit, but the Marlin-2.0 firmware
stitched directly via uart does not work without this bootloader file.bin

DroneMang, 2020-04-10:
This still does not work for me.  If I run Test it says it fails, but if I try to flash the firmware anyways, it tries but says:

waiting for sd card to mount on host

I do see a USB device mounted though with these commands:

pi@octopi:~ $ ls -al /media/
total 40
drwxr-xr-x 10 root root 4096 Dec  9 15:52 .
drwxr-xr-x 21 root root 4096 Nov 13  2018 ..
lrwxrwxrwx  1 root root    4 Dec  9 15:52 usb -> usb0
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb0
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb1
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb2
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb3
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb4
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb5
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb6
drwxr-xr-x  2 root root 4096 Dec  9 15:52 usb7
pi@octopi:~ $ ls -al /media/usb/
total 8
drwxr-xr-x  2 root root 4096 Dec  9 15:52 .
drwxr-xr-x 10 root root 4096 Dec  9 15:52 ..
pi@octopi:~ $

benlye, 2020-04-13:
That output does not prove that anything is mounted.  You need the output from the `mount` command to show that.

BastR, 2020-04-15:
This PR add support for onboard SD card. 

I’ve tried yesterday and it works, today i‘ll try with firmwareupdater.

https://github.com/MarlinFirmware/Marlin/pull/17531

BastR, 2020-04-15:
It doesn't work and my knowledge is limited, I don't know where to look.

The SD card is detected and mounted on the octoprint interface, octoprint lists the files correctly in the terminal, I can upload a gcode to the SD and print it from octoprint, but I can't mount the SD card to use the LPC1768 boards method for flashing.

NovaViper, 2020-08-02:
I can confirm this isn't working, I've followed all of the intructions for the LCP boards correctly and it still doesn't work

BastR, 2020-08-02:
There is new plugin that could replace this one for SKR PRO at least:

https://github.com/charleswillis3/OctoPrint-MarlinBft

I've installed it but not tested yet

NovaViper, 2020-08-03:
@BastR Just came that plugin a test, and it seems to be working well!

roman-karpovich, 2020-10-24:
had same issue, was able to connect after manual unmount sd card (M22)

closed 2021-01-31 by github-actions[bot]