OctoPrint/OctoPrint-FirmwareUpdater #115: Flashing failed, not in sync: resp=0x65

eroji, 2020-01-12:
I followed the instructions on configuring and flashing my Ender 5, however, I cannot get it to work. My OctoPrint is running on a pi3b+ with latest Firmware Updater plugin. Avrdude version 6.3-20171130. Serial port is /dev/ttyUSB0. AVR MCU 1284p, Programmer type arduino. Everything else is default. Select TH3D firmware, and flash. The following error is shown. I also tried all options for the baud rate, but still yields same error.

```
2020-01-12 11:26:11,287 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -V -U flash:w:/tmp/tmpztFufS:i' in /usr/bin
2020-01-12 11:26:16,696 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude says: 'not in sync: resp=0x65
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x63
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x68
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x6f
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x3a
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x53
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x44
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x69
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x6e

avrdude done.  Thank you.'.
```

benlye, 2020-03-12:
Are you able to flash the board using any other method - e.g. directly connected to a PC using Arduino IDE?

eroji, 2020-03-12:
Yes, using PC and IDE works. However, I no longer own this printer.

closed 2020-03-12 by benlye