OctoPrint/OctoPrint-FirmwareUpdater #122: Can't flash Creality v1.1.5 board

LipSanFlip, 2020-02-26:
**Hardware Setup**
Ender-3 Pro, Creality v1.1.5 silent board, RP4 running Octorprint 1.3.12

**Describe the problem**
I've been trying to flash new firmware and continue to get errors. Please see the log below:

**Log Files**
[***** FIRMWARE UPDATER LOG *********]
2020-02-26 20:52:50,974 /usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpXbnDVC:i
2020-02-26 20:52:51,560 > 
2020-02-26 20:52:51,561 > avrdude: Version 6.3-20171130
2020-02-26 20:52:51,561 >          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
2020-02-26 20:52:51,561 >          Copyright (c) 2007-2014 Joerg Wunsch
2020-02-26 20:52:51,562 > 
2020-02-26 20:52:51,562 >          System wide configuration file is "/etc/avrdude.conf"
2020-02-26 20:52:51,562 >          User configuration file is "/home/pi/.avrduderc"
2020-02-26 20:52:51,563 >          User configuration file does not exist or is not a regular file, skipping
2020-02-26 20:52:51,563 > 
2020-02-26 20:52:51,563 >          Using Port                    : /dev/ttyUSB0
2020-02-26 20:52:51,564 >          Using Programmer              : arduino
2020-02-26 20:52:51,564 > 
2020-02-26 20:52:57,632 > avrdude: stk500_recv(): programmer is not responding
2020-02-26 20:52:57,633 > avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

[***** OCTOPRINT LOG *****]
020-02-26 20:52:57,841 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.9 (Github)"
2020-02-26 20:52:57,862 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 20:52:57,892 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 20:53:00,042 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 20:53:04,344 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 115200, u'printer_port': '/dev/ttyUSB0', 'firmware_name': 'Marlin 1.1.9 (Github)'}
2020-02-26 21:02:19,836 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-02-26 21:02:29,830 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 904}
2020-02-26 21:02:46,084 - octoprint.plugins.firmwareupdater - INFO - Firmware update started
2020-02-26 21:02:46,092 - octoprint.plugins.firmwareupdater - INFO - Disconnecting from printer
2020-02-26 21:02:46,109 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Offline"
2020-02-26 21:02:46,126 - octoprint.plugins.firmwareupdater - INFO - Running '/usr/bin/avrdude -v -q -p m1284p -c arduino -P /dev/ttyUSB0 -D -U flash:w:/tmp/tmpBGD0Vm:i' in /usr/bin
2020-02-26 21:02:46,279 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.1.39) 11.14ms
2020-02-26 21:02:46,287 - tornado.access - WARNING - 409 POST /api/printer/command (::ffff:192.168.1.39) 17.83ms
2020-02-26 21:02:52,843 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Avrdude says: 'not in sync: resp=0x00'.
2020-02-26 21:02:52,847 - octoprint.plugins.firmwareupdater - INFO - Reconnecting to printer: port=/dev/ttyUSB0, baudrate=115200, profile={'heatedChamber': False, 'name': 'Ender 3 Pro', 'color': 'default', 'axes': {'y': {'speed': 6000, 'inverted': False}, 'x': {'speed': 6000, 'inverted': False}, 'z': {'speed': 200, 'inverted': False}, 'e': {'speed': 100, 'inverted': False}}, 'heatedBed': True, 'volume': {'origin': 'lowerleft', 'formFactor': 'rectangular', 'depth': 220.0, 'width': 220.0, 'custom_box': False, 'height': 250.0}, 'model': 'Ender 3 Pro', 'id': '_default', 'extruder': {'count': 1, 'nozzleDiameter': 0.4, 'offsets': [(0.0, 0.0)], 'sharedNozzle': False}}
2020-02-26 21:02:52,937 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2020-02-26 21:02:52,944 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial port" to "Connecting"
2020-02-26 21:02:52,953 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-02-26 21:02:52,955 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-02-26 21:02:52,964 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2020-02-26 21:02:52,967 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-02-26 21:02:52,970 - octoprint.plugins.firmwareupdater - INFO - Got CONNECTED event
2020-02-26 21:02:52,974 - octoprint.plugins.firmwareupdater - INFO - Run postflash flag is not set
2020-02-26 21:02:53,003 - octoprint.util.comm - INFO - Got a resend request from the printer: requested line = 1, current line = 3
| Last lines in terminal:
| Connecting to: /dev/ttyUSB0
| Changing monitoring state from "Offline" to "Opening serial port"
| Connected to: Serial<id=0xa885ac10, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
| Changing monitoring state from "Opening serial port" to "Connecting"
| Send: N0 M110 N0*125
| Send: N0 M110 N0*125
| Recv: echo:Unknown command: "0 0 0 0 N0 M110 N0"
| Recv: ok
| Changing monitoring state from "Connecting" to "Operational"
| Recv: ok
| Send: N0 M110 N0*125
| Send: N1 M115*39
| Recv: ok
| Send: N2 M21*18
| Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
| Recv: Resend: 1
2020-02-26 21:02:53,028 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 1.1.9 (Github)"
2020-02-26 21:02:53,043 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 21:02:53,082 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 21:02:55,244 - octoprint.util.comm - INFO - Firmware states that it supports temperature autoreporting
2020-02-26 21:02:59,551 - octoprint.plugins.tracking - INFO - Sent tracking event printer_connected, payload: {u'printer_baudrate': 115200, u'printer_port': '/dev/ttyUSB0', 'firmware_name': 'Marlin 1.1.9 (Github)'}

**Additional context**
Thanks for your help!

benlye, 2020-03-12:
Are you able to flash the board using any other method - e.g. directly connected to a PC using Arduino IDE?

benlye, 2020-04-13:
No response.

closed 2020-04-13 by benlye